arendst/Tasmota #598: MESSAGE: Change default flash mode to DOUT - Request for Comment

arendst, 2017-07-11:
As more new devices are released using the esp8285 chip it becomes important to initially load firmware compiled with the correct flash mode of DOUT.

While the default flash mode for esp8266 is DIO/QIO I've done some testing using DOUT on all my esp8266 devices and it seems to work just fine. There might be some theoretical delay in flash access (from single to quad access) but I did not notice any delay during operations or while flashing new firmware or saving settings.

Upgrading current devices to DOUT compiled firmware is flawless. Initially they will still be using DIO/QIO but future upgrades will change them to DOUT. Using command ``Flashmode 3`` sets DOUT format but this will only stay active after I've released the proposed change as current versions change flash mode according to selected module during upgrades.
 
So I plan to release new firmware only compiled with flash mode option DOUT. The wiki and the ``platformio.ini`` file will be updated accordingly and the binary ``sonoff-touch-4ch-bnsz.bin`` will be obsolete.

Feel free to comment and let me know if there might be an issue I haven't anticipated.

arendst, 2017-07-14:
Will be introduced with next release

ileez, 2017-07-20:
Thanks, has been resolved.

closed 2017-07-22 by arendst