arendst/Tasmota #603: How to flash wemo d1 mini

johannesbonn, 2017-07-12:
Hi,
I have tried several times to flash a wemo d1 mini with platformio connected by a ftdi (set to 5V). Upload process seems successfull but connected to the USB port for regular power supply nothing happens. I can't find the wemos in my wlan (settings are set in the config file). Is my way for flashing right?
Can anybody explain me how to flash the wemos?

Thank you!

Courty40, 2017-07-12:
i've not tried it with platformio but flashed one only yesterday using the Arduino IDE and ESP8266 addins.
Once flashed, it came up straight away and worked perfectly.
an external FTDI should be set to 3.3v really but why not use the onboard FTDI USB connection ?

kruebsam, 2017-07-12:
I am using platform IO with various devices (although yet no wemo one) and so far never had any problems. As already mentioned by @Courty40 be aware of proper power on the serial lines. 5V will quite likely destroy the RX-port of your ESP8266.

johannesbonn, 2017-07-12:
Ok, thank you. By using the USB port for flashing, how I can set the wemos in flashing mode?

Courty40, 2017-07-12:
on the Wimo's its does it automatic as the upload starts

johannesbonn, 2017-07-16:
Thank you for your help. My first Wemos was destroyed by the 5V. I bought a new one and now I stug again. I tried flashing over the usb port of the wemos with this result:

`Looking for upload port...
Auto-detected: /dev/cu.wchusbserial410
Uploading .pioenvs/sonoff/firmware.bin

error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
error: failed reading byte
warning: espcomm_send_command: cant receive slip payload data
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error 255`

I am using mac os sierra and platformio. I have installed the newest driver without the kernel panic problem.
Can somebody point my in the right direction? I don't know what to do....

johannesbonn, 2017-07-17:
Ok, flashing seems successful when I changed in platformio.ini variable board = d1_mini. But the mini didn't react after restarting. No activity or anything else. Is there any other clue I have to consider?

johannesbonn, 2017-07-18:
I started from scratch today and now everything is fine. I don't know why, but thanks for this great software!

closed 2017-07-18 by johannesbonn

demian85, 2018-06-10:
My Mac does not seem to recognize the Wemos D1 connected via USB. Do I need a driver or what? Should I use serial? Wiki does not specify which connection to use, it assumes I'm using serial, but the board has an USB port.. Thanks.

pasqui7, 2018-07-12:
Hi guys, can you tell me how can I put the device wemos d1 mini : 

https://www.amazon.it/gp/product/B075HC84N8/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1

into flash mode? 

I need ft232rl device to flash it, or I can only  install software driver and connect directly on my computer the device with micro usb cable?

thank you