arendst/Tasmota #604: Simple push-button for Sonoff SV?

jstorgaard, 2017-07-12:
I want to install a Sonoff SV for opening my garageport. I got 24V output for the SV and the garageport motor got input for connecting a simple push-button for triggering open and close of the port.
How should I configure the SV for emulating push of a button?

ionciubotaru, 2017-07-12:
use GPIO 0 - is configured by default to toggle the relay, or better read https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic

GhsHun, 2017-07-13:
Use PulseTime Command. (for example PulseTime 3 = 0.3 sec impulse) See it: [https://github.com/arendst/Sonoff-Tasmota/wiki/Commands](url) After that you send a Power On command, the switch will be On state and Off state immediately and send an impulse to the motor. This impuilse will open or close the door. That's how I solved the thing.

jstorgaard, 2017-07-16:
Thank you for your replies. 
I will have a go with the PulseTime command.
For the output - will I use the regular outputs or some other outputs on the SV board for the GPIO?

closed 2018-05-07 by stale[bot]