arendst/Tasmota #605: Inverting ON/OFF Leds on Sonoff Touch

PeggyFree, 2017-07-13:
**Currently :** 
Relay OFF : button LED is OFF, wifi LED is ON
Relay ON :  button LED is ON, wifi LED is OFF

All my family and guests get trapped : whend the living room is dark, they try to push the wifi button which is enlightened, then they get confused because nothing happens.

**Expected:**
Relay OFF : button LED is ON, wifi LED is OFF
Relay ON :  button LED is OFF, wifi LED is ON

Like this the Touch would be more user friendly and behave like my other touch switches : livolo, neo coolcam.

zizebra, 2017-07-13:
I have noted this behavior on my new esp8285 one channel. It does not happen all the time. It does happen after a while. However I am not sure if this is related to the issue which you are experiencing

lxz81, 2017-07-13:
I've seen this in another issue: https://github.com/arendst/Sonoff-Tasmota/issues/398#issuecomment-298324476

Seems what you expect is not possible

davidelang, 2017-07-13:
according to the command page, the led state is inverted for the touch, what 
version of the firmware are you running?

see ledstate

https://github.com/arendst/Sonoff-Tasmota/wiki/Commands
k

PeggyFree, 2017-07-14:
I'm using the 5.1.5. As far as I know the ledstate is mapped on the wifi signal and the push button led is inverted with this. 

The answer to my question is given by arendst in the #398 : The touch has two leds. One connected to the same control line as the relay (in the middle) so that when the relay is on the led is also on. Unable to control it seperately from the relay.

Looks like nobody can change this behaviour, unfortunately. This is a bad design from ITEAD, IMHO.

closed 2017-07-14 by PeggyFree

glsf91, 2019-03-16:
@PeggyFree 
See my solution in #398 .
Now it is always on.