arendst/Tasmota #606: Add additional button/switch to Sonoff 4CH on GPIO

untitledlt, 2017-07-13:
I'm using Sonoff 4CH and I would like to add additional switch or push button so I could get it's state or change over MQTT. Is it possible to set free GPIO pin as INPUT mode and get it's state over MQTT?

davidelang, 2017-07-13:
yes, you can define a switch and use switchtopic

untitledlt, 2017-07-13:
but i can only select 1-4 and all of them are already used on Sonoff 4CH for built-in buttons

davidelang, 2017-07-13:
buttons 1-4 are in use, switch1-4 are not, they are not the same thing. define a 
switch and set the switchtopic and the buttons will keep working as-is and the 
switches will send a message with a different topic.

untitledlt, 2017-07-13:
oh cool, it's working but it would be great if it could report ON/OFF state instead of TOGGLE. Am i missing something here?

davidelang, 2017-07-13:
look at switchmode

untitledlt, 2017-07-13:
Thanks! It works now. Before i changed `switchtopic` it was using the same topic and toggling my relays so that's why i thought that Switch1 was the same as Button1 :)

itech-dean, 2017-08-10:
Sorry to high jack this post. I am trying to do the same thing. Can I use J5 to achieve this? 

Cheers

Rich

dawsmac, 2017-08-10:
Can someone please help me with this I am trying to achieve the same but need to understand if I can use J5 on the board?

kajmaj, 2018-03-16:
As I learned here, Use J1 RX/TX as they are GPIO 1/3.
You must disable Serial Logging before you will use it.