arendst/Tasmota #607: Still can't get my Sonoff flashed (esp_comm sync failed). Help!

EttVenter, 2017-07-16:
Hey guys!

After watching the video on Youtube, and going through all the comments to see the issues everyone else is having with:

```
Uploading .pioenvs/sonoff/firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
```

I see that the issue is FTDI related, and that people seem to not be getting the Sonoff in "program mode" or whatever it is. I've tried everything and I still can't get it working. I've connected the FTDI first, and THEN the Sonoff even, and that also doesn't work. 

Just can't get this flashed. Any ideas? BTW, the FTDI I have is the itead Foca v2.2

Thanks!

sujitrp, 2017-07-17:
I was having this issue till last week . My sonoff were dead was not responding . I was about to scrap project .After 2 weeks of trial and error my conclusion is some USB-TTL does not works due to drivers issue or connection issue . 
I used UNO to flash Sonoff .
https://www.youtube.com/watch?v=qqSLwK1DP8Q
add GND pin for sonoff  and It is worked. My dead sonoff are now working fine.

robinvalk, 2018-01-14:
@sujitrp Dude, I think you just saved me! Spend 2.5 hours trying with a CH340G and an external power supply. This worked in 5 minutes. Thanks!!

closed 2018-05-28 by stale[bot]