arendst/Tasmota #608: For those having issues flashing the Sonoff Basic

mihalski, 2017-07-17:
This did not work for the Sonoff Pow, but I was able to flash those via FTDP even for the latest units without issues.

https://github.com/mirko/SonOTA/issues/10#issuecomment-315682511

Good luck.

DarkAllMan, 2017-07-17:
I'm having problems doing this.... can you help me?
I get an error running the script: Following arguments are required: serving_host

mihalski, 2017-07-17:
That's because you're missing the 3rd argument, serving_host. You need to provide the IP address of the machine you are running the script on and it needs to be accessible on the Wifi network that you provided.

empossible, 2017-07-18:
When I run with "python" I receive this:

>  File "sonota.py", line 397
    print(".", end="", flush=True)
                         ^
SyntaxError: invalid syntax

When I run with "python3" I get this instead:

>Traceback (most recent call last):
  File "sonota.py", line 24, in <module>
    import tornado.ioloop
ModuleNotFoundError: No module named 'tornado'

Also, should the Sonoff be in flash mode, or just plug it in for power and data transfer?

Thanks for the help.  My first one went fine, these two new ones don't seem interested in anything.

mihalski, 2017-07-18:
You haven't installed the python requirements for SonOTA as mentioned.
And neither. As described you power the Sonoff so that it slowly flashes the light and then you put it into the ITEAD flash mode by holding the button until it flashes quickly. Exactly as you would if updating the ITEAD firmware.

When you say your first one went fine do you mean with this method? What types of Sonoff are these?

DarkAllMan, 2017-07-18:
It seems I have succeeded to flash...... the Sonof is blinking steadily.
It seems to be getting an ip address, however, I'm not able to connect to the webpage... what could be going on? Is the webpage on a special port?

empossible, 2017-07-18:
Whoops, didn't realize your comment linked to SonOTA; I've now installed all the prerequisites.

All three are Sonoff Basics, the first one I followed Bruh's directions on YouTube and installed 5.2.3 via the Arduino app.  I tried that with the second two and that didn't work.  Then I tried with QuickPi's directions and installed via Atom with PlatformIO.  I've tried 5.2.3, 5.3.0, and something in the 4.x.x range.  I've even used the esptool to erase the firmware.  Each time I did not get any wireless connection.  I'm running on macOS 10.11.6, by the way.

Now with the proper prerequisites installed and your directions (including creating the indebuurt1 WiFi network), I get this:

> python3 sonota.py --wifi-ssid indebuurt1 --wifi-password VnsqrtnrsddbrN 10.0.1.13
** No ip address of the ITEAD DHCP range (10.10.7.0/24) is assigned to any of your interfaces, which means you don't appear to be connected to the IEAD WiFi network.
** Please change into the ITEAD WiFi network (ITEAD-100001XXXX)
** This application can be kept running.
............. (just a hole lot of repeating dots here)

No lights on the Sonoff either.  If you don't have any other suggestions, I'll try again on my Windows 10 machine at work tomorrow.

mihalski, 2017-07-18:
As it says you don't appear to be joining the ITEAD wifi network after you run the script as per my instructions. If the Sonoffs no longer have the original firmware on them you will no longer be able to use this method.

DarkAllMan, 2017-07-19:
Any comment on my situation? Do you know the led status?

mihalski, 2017-07-19:
@DarkAllMan As I understanding it slow blinking is normal (for unconfigured Sonoff running ITEAD firmware and fast blinking is when the ITEAD firmware goes into programming mode).

If you currently have the original Sonoff firmware you should be able to connect to the device on an ITEAD wifi network if you hold the button until it starts flashing quickly again.

DarkAllMan, 2017-07-19:
@mihalski https://www.youtube.com/watch?v=CkqOqYJ9FZM
Sorry about the VSS

mihalski, 2017-07-19:
Not pushing.. Holding.

DarkAllMan, 2017-07-19:
Let me be clear, I managed to do the OTA update..... so there is no more itead firmware.
Holding the pin does not change the status of the LED..... I held it, 5 seconds, 10 seconds, 20 seconds....

Lennyz1988, 2017-07-24:
@DarkAllMan I have no idea what you are trying to show us with your Youtube link. What are you trying to show us?

DarkAllMan, 2017-07-24:
I’m trying to show you the LED. I don’t know what the LED status means.
First it start blinking fast, then medium fast. It does not give me an Itead WIFI. It does not connect to indebuurt1. I cannot flash it via FDTI.
I have ordered a new FTDI to test if it will work……

teranus, 2017-10-30:
I understand you, i have the exact same blinking behaviour after OTA with my Sonoff 4ch.
When I switch on the power, it flashes ca 5sec very quick and afterwards just quick - like in your video.
I found no way out.
Holding the button for 7sec or even more - no effect
Pressing the button 3times quickly - no effect
The same with 4 times or 5 times
I even tested these during the first very quick flashing periode and in the second one...
Would be highly interested in a solution for that.

With the same OTA methode I succesfuly flashed a Sonoff Basic.
Here I now play around integrating it with MQTT broker in FHEM (as I'm still new to all of this)

Unfortunately I have 2 more devices where the OTA not worked out 'properly'...
It is a second, equal Sonoff Basic and one Sonoff S20.
When I give them power, they don't connect to a Wifi
Not indebuurt1 (hotspot on my mobile) and not my home-network
The button pressed once toggles the relais correct.
The button pressed 4 times lets LED blink and after short I have a ITEAD-.... Wifi to connect. Then a configpage looking like Tasmota is opening on 192.168.4.1 (automatically) and ask me again for AP1 and AP2. Equal what I fill in for AP1 or AP2 it is not connected afterwards.
The only option beside safe is to just reboot. Is there in THIS state some possibility to flash the device instead of wificonfig and/or rebooting?
Or am I able to reset it totally and start with something default?

closed 2018-05-07 by stale[bot]