arendst/Tasmota #610: Update GroupTopic <XYZ Room> is resulted <XYZ_Room>

Sunrise17, 2017-07-17:
Dear @arendst ,
Thanks for the useful update (5.3.0), Backlog works so well. I have one issue after update. Before 5.3.0, i was using v.5.1.3 and my group topic could accept space between words. However, with 5.3.0, its adding "_" to spaces between words. I have checked same thing in FriendlyName, there is no such problem in that.

[image]

Greetings
Koray

arendst, 2017-07-17:
Good point. I changed it thinking that mqtt would not allow spaces in topics. Apperently mqtt does allow spaces in topics.

I have to check what the consequences are for my code if i reallow spaces ...

Sunrise17, 2017-07-18:
It will be so perfect, if you can reallow unless it will be any problem for your code.
Thanks for your support,

Sunrise17, 2017-08-01:
Dear @arendst,
I am closing this issue, there is no need to use spaces between words in topic in Mqtt. I have solved the issue. Thanks for your attention,

closed 2017-08-01 by Sunrise17