arendst/Tasmota #612: 1 Ch Sonoff Relay State on Startup

zizebra, 2017-07-17:
Hi All 

I have an issue on my 1 channel off Relay which is based on Esp8285.  I have noted that the relay ordinarily works when everything is ok. However if the device reboot due to an error or exception, then the relay is activated during for 1 second during the boot phase. 

This is a problem me as i have this 1 ch relay connected to my garage door and every time it reboots unexpectedly then the door is opened.

electrontube, 2017-07-18:
I had the same issue here when I flashed a brand-new module with release 5.3.0
on other modules with 5.1.2 there was no issue
so I tried 5.1.2 on the new module that was causing the problem with 5.3.0 and guess what... 
...it worked fine
next I tried 5.2.1 --> working fine
then 5.2.4 --> working fine

I then upgraded to 5.3.0 and it works now fine as well.
this sounds strange but it's true

since I don't have any new modules left over, I can't reproduce it
probably the procedure above helps to solve the problem for you as well.

however, since several people seem to have this issue, 
Theo will have to have a closer look to find out what causes this behavior

zizebra, 2017-07-18:
Thank you. I will try this.

zizebra, 2017-07-19:
so far so good with  version 5.2.1. May be its because it hasnt crashed or restarted unexpectedly since yesterday.

electrontube, 2017-07-19:
On the 5.2.1 release - have you configured everything as it was on 5.3.0 (especially topic name)?
I was able to reproduce the error by changing the topic name

zizebra, 2017-07-20:
My topic is the same as it was on 5.3.0 but that topic is not the default. It is cmnd/gate/power. 

The  challenge so far is that the device has been running close to 35 hours without unexpectedly restarting. Hence its difficult to say that the issue isn't there. On 5.3.0 the device crashed and restarted once to twice or more per hour. The first 8 hours was fine though but once it restarted the crashing, that would open the gate. So on this version 5.2.1 I  would like to see what happens if it restarts which has not happened.

zizebra, 2017-07-21:
Its been running for 62 hours without crashing.  Therefore my gate has working properly. Still cant tell if the relay will activate when it crashes next time.

electrontube, 2017-07-21:
So we still don't know what the actual cause was (or still is).

zizebra, 2017-07-21:
We wont know just yet, unless it reboots. Iam waiting for another similar device which i will use with my pedestrian gate. I will have time to play with it to investigate this.

zizebra, 2017-07-23:
So today someone switched off the power to the esp8266 relay and immediately switched it back on which caused it to restart. During the restart, the door opened. I tried to reproduce it by switching off and back on. Could not reproduce it

zizebra, 2017-07-23:
Please ignore my earlier comment. The power loss could not have opened the door. Remote was used to open the door almost same time when the power for door was switched off

zizebra, 2017-07-25:
Ok, After running for more than week excluding the reboot on Saturday, I can confirm now. The esp crashed and rebooted a few minutes ago and door was opened.

zizebra, 2017-08-03:
I guess the 1ch sonoff is not usable for controlling doors. I have noted thus far that blocked loop reboots the device 48-62 hours. Whenever it reboots with blocked loop, the it powers on momentarily. I guess I have to make my own 1ch esp relay. This sonoff should be used with lights.

electrontube, 2017-08-12:
Since it appears to be stable in my "Lab" I have decided to go live with 5.5.x
I have a "Feedback" Switch on GPIO14 tag will trigger an E-Mail as soon as the Door leaves it's "closed" Position.

Stay tuned. . .

ratedz, 2017-11-25:
I have this exact problem its killing me. Did anyone figure out how to get around this yet ?

zizebra, 2017-12-13:
Based on this guys work, http://rabbithole.wwwdotorg.org/2017/03/28/esp8266-gpio.html
That wont go away unless you make changes to the hardware. GPIO 4 and 5 are the only ones which appears not to go high during reboot/reset.  Itead 1ch  relay uses GPIO12 and hence problem will always be there..

rsdatatechnologies, 2018-01-21:
Hi Guys, I too had this issue or a similar one. I realised that my issue was actually caused by the fact that homeassistant was using the retain flag on the commands sent to sonoff. So when sonoff starts up it followed the commands saved. To solve the problem I simply sent an empty byte on the topic to clear to the retained message, then made sure no more commands used the retain feature.

guozz1984, 2018-01-26:
@rsdatatechnologies,

If I am right;

I have to set in configuration: “retain: false”
And sent another “cmnd/sonoffname/nothing”

???

rsdatatechnologies, 2018-01-27:
@guozz1984 Yes, you are correct. Just send an MQTT publish with the rain flag and an empty message to remove.

Something like this would work:

mosquitto_pub -d -u <user> -P <password> -t cmnd/sonoff/desklamp/power -m "" -r

Also, you can see all retained messages by just subscribing to all topics with "#".

mosquitto_sub -d -u <user> -P <password> -t "#"

rmarkle, 2018-05-02:
@rsdatatechnologies Can you send a rain flag through home assistant?  I'm not sure how to do that

rsdatatechnologies, 2018-05-02:
@rmarkle 

Apologies I meant "retain" flag, will edit it now.

I'm not sure, the easiest way is to install mosquitto and mosquitto_clients 

sudo apt-add-repository ppa:mosquitto-dev/mosquitto-ppa
sudo apt-get update
sudo apt-get install mosquitto
sudo apt-get install mosquitto-clients

Then use the commands listed above.

closed 2018-07-01 by stale[bot]