arendst/Tasmota #613: MQTT Connect FAILED

EmilySmith0, 2017-07-17:
Hi Guys I am getting MQTT connection issue with my Sonoff basic Relay
I want to integrate my sonoff with HomeSeer using MQTT protocol
My homeseer and Mosquitto broker is on same pc with IP 192.168.1.102
Homeseer plugins are able to communicate with Mosquitto broker but sonoff showing following error

00:00:17 MQTT: Attempting connection...
00:00:18 mDNS: Query done with 0 mqtt services found
00:00:24 MQTT: Connect FAILED to 192.168.1.102:1883, rc -2. Retry in 10 seconds
Thanks
Emily
[image]

khcnz, 2017-07-17:
The error codes (-2) is documented here https://pubsubclient.knolleary.net/api.html#state indicates that it can't connect to that ip address.

EmilySmith0, 2017-07-17:
I don't know what to do to establish connection
Here is MQTT configurations image
[image]

EmilySmith0, 2017-07-17:
I allowed mosquitto broker in firewall and all is working now
First click add another program in firewall settings and browse for mosquitto and tick mark to allow in home and public networks

closed 2017-07-17 by EmilySmith0