arendst/Tasmota #616: Wifi Selection

Lawrencezarb, 2017-07-19:
I have 2 wifi networks setup on my Sonoff basic, but when it boots up it does not select the stronger wifi signal. The stronger wifi signal is configured on AP1 and the weaker on AP2, but the Sonoff always seems to select AP2.

The only way I can get it to connect to AP1 is to change the SSId on AP2 so it cannot connect at all to AP2.

davidelang, 2017-07-19:
it doesn't try to pick the 'best' AP, it just tries each of them until one 
succeeds. I don't know why it's failing to connect to the first one, but it 
could be a matter of timing.

Lawrencezarb, 2017-07-19:
Thanks David, is there a way to manual force it to AP1 via the console for example?

davidelang, 2017-07-19:
yes, see the commands page of the wiki

closed 2018-05-07 by stale[bot]