arendst/Tasmota #618: Webserver port setting

Lawrencezarb, 2017-07-19:
When I initially flashed Tasmoto 5.2.2 I changed the Webserver port in user_config.h to 2525.

After upgrading to 5.3.0 I find that the webserver port has defaulted to 80. Can I change the webserver port from the console or do I have to amend user_config.h and reflash 5.3.0 with the FTDI cable again?

zizebra, 2017-07-21:
I think you can change the port to your desired number in the code, compile and export a new version. You then upload the new version via web. That should work.

arendst, 2017-07-22:
Web server port is hardcoded using user_config.h

In your case you will have to recompile with your required web port as @zizebra suggested.

Lawrencezarb, 2017-07-22:
Thanks, that worked for me

closed 2018-05-07 by stale[bot]