arendst/Tasmota #619: Sonoff Pow bricked after OTA upgrade from 5.1.3 to 5.3

jkrogsgaard, 2017-07-19:
I upgraded from 5.1.3 to 5.3 and then my Pow never came online again. It's very hard for me to unmount the Sonoff from where it is now, so I really hope that there's a "easy" way to get it online again.

I tried to push the button 6 times for reboot, that didn't help.
I tried to hold the button for more than 4 seconds for reset, but that didn't help either.

Right now it's off line and just blinking blue once every 8 seconds.

What to do?

ionciubotaru, 2017-07-20:
You must rewrite the flash it via serial.
I had the same problem, I succeed only with manual upload via web interface
The problem appears at POW only, other devices are upgrading fine.

jkrogsgaard, 2017-07-20:
:-/ Damn - It's placed so specific that this will be a very hard exercise. Is everybody sure that there's no other way to do it?

zizebra, 2017-07-21:
Iam so sure, if it is offline, then you are need to crawl in there and get flashed serially again. it is times like that makes me realise the importance of backup partition which can be used to bootup the device when it is failed like this. I hope the makers of the hardware realise this sooner.

davidelang, 2017-07-21:
On Fri, 21 Jul 2017, zizebra wrote:

> it is times like that makes me realise the importance of backup partition 
> which can be used to bootup the device when it is failed like this. I hope the 
> makers of the hardware realise this sooner.

That requires twice the storage, and the manufacturers aren't interested in 
providing more storage for the convienience of people who are going to make the 
devices NOT connect to their cloud service.

closed 2018-05-07 by stale[bot]