arendst/Tasmota #624: Power state not being saved

maxdamage2122, 2017-07-20:
Not sure if this is a setup issue or a bug.
I have a sonoff basic setup with mqtt and using home assistant. If I toggle the sonoff via HA and reboot it through the webserver, it boots back up in the correct previous power state. But if I toggle the sonoff via the webserver or through a wired switch and reboot it then it boots back up with the previous state set by HA.

davidelang, 2017-07-20:
read the poweron state page of the wiki

https://github.com/arendst/Sonoff-Tasmota/wiki/PowerOnState-Configuration

maxdamage2122, 2017-07-21:
I had already read that page but I think I found my issue. With switchtopic on, the topic uses POWER1 and in my HA config I was using POWER. Switching my HA config to use POWER1 seems to have fixed the issue.

closed 2017-07-22 by arendst