arendst/Tasmota #626: Boot loop when i change parameters and save

zizebra, 2017-07-21:
On my other device, using Sonoff Basic template, iam having an issue where the device fails as soon as i assign anything to the GPIOs. I have tried assigning Relay2, Switch2, LED1 to any of the GPIOs. As soon as i save, the device reboots and fails. The only way i can recover the device is through complete erase of the flash. Then device works fine as long as i do not assign anything to the GPIOs. I am trying to use this device as a switch and buzzer.

So with this device i have two issues, why cant this device operate normally and allow me to use GPIOs, why is this failing?

Secondly, I am trying to use the blinker for buzzing. Currently the blinker seems to blink for 1 sec. I would like to shorten the blink time, is this possible? I changed blinktime but i didnt notice any change on the actual blink time

canis-ferrin, 2017-07-22:
Confirm.

arendst, 2017-07-22:
If I use a Sonoff Basic Module with default one relay and button and add a relay2 to GPIO2 ir reboots as expected and returns two relays in the web interface as expected.

[image]

 
Please provide step-by-step procedure leading to your failure as I cannot reproduce. 

Start with ``status 0``.

ionciubotaru, 2017-07-23:
I flash more than 50 sonoffs (touch, pow, basic, electrodragon) with Tasmota. I found this issue only on one  [BN-SZ01 WiFi Round Flush Mount LED Ceiling Light](https://www.itead.cc/bn-sz01.html) and I think it's a hardware not a software issue. Unfortunately I have only one lamp so cannot told you if all lamps has this problem or only mine.

zizebra, 2017-07-23:
Hi theo
that is exactly the process I have used. Assigning anything to any of the GPIOs renders the device inoperable when  it restarts. 

I have flashed the same device with easyesp firmware and it's working. It could be a hardware issue,  I don't know. But it is working with an alternative firmware.

closed 2018-05-07 by stale[bot]