arendst/Tasmota #627: FYI: Sonoff S22 Compatible

tloten, 2017-07-22:
I noticed the Sonoff S22 isn't listed as compatible on the main Readme.md, but I can confirm it works fine after following the QuickPi tutorial video.

This is the unit I used - seems to be the more commonly available ones with Australian/NZ plug:
https://www.banggood.com/SONOFF-S22-10A2200W-AU-Plug-Wireless-Remote-Control-Smart-Socket-Phone-App-Control-Wifi-Smart-Home-p-1142284.html

Info after uploading Sonoff-Tasmota:
```ESP Chip id	15304653
Flash Chip id	1327328
Flash size	1024kB
Program flash size	1024kB
Program size	474kB
Free program space	528kB
Free memory	27kB
```

I can't find any S22 info on the itead website though, perhaps it's some kind of knockoff? Everything was branded as Sonoff and looked legitimate. 

Anyway, just thought this may be useful info in case you wanted to list it as a compatible device.

arendst, 2017-07-22:
Thnx for the info. It looks to me as it is a combination of S20 and a TH10 using the AM2301 sensor. 

Do you use the S20 Socket or the Sonoff TH module type?

tloten, 2017-07-22:
Yeah - the optional temperator/humidity sensor is the AM2301.
I initially had it configured as the `01 Sonoff Basic` module type and it worked fine with that - even the temp probe was working.
Just out of curiosity I just switched it to the TH module type and that's working fine too.

Many thanks for the project - it's nice when it all works first time & is so well documented :)

jat80, 2017-09-18:
@tloten Do you have any pic's of the inside of the s22?  Was it easy to open to flash it?

tloten, 2017-09-19:
Sorry no pictures. I recall it was a single screw to open it, and all the programming pins were labelled on the PCB. There was plenty of room to solder in pin headers to make future re-programming easy.

jat80, 2017-09-19:
@tloten - perfect thanks!  I bought the AU version of the S20 and it didn't have much room to solder on header pins. still managed to flash it though. 

I'll grab an S22 off AliExpress and give it a go. Thanks again.

CliveBull, 2017-10-20:
@tloten @jat80 @arendst I confirm the S22 works fine here in New Zealand with the AM2301 plugged in. Two inside tabs at the bottom and two screws at the top. Remove screws, tear the warranty sticker and it pops open easily. Some S22 photos below.

I soldered on a 4-pin header and flashed it with 5.8.0. No issues, wifi came straight up. Configured it as a TH and it reads the temp and humidity and publishes correct MQTT. So S22 and TH appear to be 100% synonyms as far as I can see. 

[image]
[image]
[image]

jat80, 2017-10-20:
Awesome!  Thanks for the info and pictures bro! @CliveBull

arendst, 2017-12-27:
Link to this issue added to Home page.

closed 2017-12-27 by arendst

QAMfan, 2022-10-02:
Recognising this is 5 years old I just noticed in Clive's pictures the S22 is wired incorrectly for AS/NZS standards. The S22 is made for domestic chinese market I suspect. In CHina the active and neutral pins are reversed compared to AS/NZS standardsa. So the relay is switching the neutral line and not the active line. Very dangerous indeed unless it is re-soldered...
[image]

CliveBull, 2022-10-02:
@tloten @jat80 @arendst - copied as interested parties.

@QAMfan - you're absolutely correct. I bought 3 of these in October 2017, not for switching 230 V but for their ability to plug and play a sensor in and get data via MQTT per Tasmota - the AM2301 you can see in the photos. I'm in NZ.

100% agree that out of the box the S22 switches the Neutral, not the Phase (live) - this is incorrect for AU/NZ use. I opened one up today and was appalled that I didn't pick this up at the time. The fix for AU/NZ is to (1) swap the live and neutral input pins: using a pair of pliers, shove them in and put them back in reversed AND (2) un-solder the two outputs and re-solder them swapped over. You have to resolder the outputs because the live and neutral sockets cannot be pushed in like the input pins. Leave the Earth as-is. That's it.

Note that Sonoff's internal wire colour choices are awful: red is OK for Live but black for Earth and green and yellow for Neutral are 100% wrong.

Re similar devices, I also have an S26, which is specifically marketed by Sonoff for AU/NZ. Same as the S22 but without the sensor socket. I opened that today as well and, while the internal wiring colours take a bit of artistic license, Live and Neutral are exactly where they should be for AU/NZ use.

Many thanks for the post! 

Clive

tloten, 2022-10-03:
Thanks QAMfan, very nice spotting. Simliar to Clive, I've been using it purely for temperature/humidity sensing too, so fortunately haven't been using the switch feature of it. I opened one up yesterday and indeed the neutral is being switched. Glad to hear that the newer S26 model is correctly wired!