arendst/Tasmota #628: After power loss, SONOFF ignores PULSETIME and stay in ON status

lobocobra, 2017-07-22:
Hi all

Hopefully some user can help me, I will document the solution path here, for others if they have the same problem. I installed version 5.3 but the problem is also in 5.2.4.

**Problem statement**
I have the regular single sonoff and would like to use it for my window stores. In consequence it must run 15 secs and then stop. BUT it is important, that it stops also after a power loss, in order to prevent, my store motors burn to death.

**Whatworks...**
    - SONOFF connects to WIFI
    - I can turn ON/OFF by MQTT / button
    - I can make it OFF automatically after 15 secs ON

**MY PROBLEM**
    - after a power loss, SONOFF goes in POWER ON mode
    - SONOFF ignores PULSETIME settings and remains ON FOREVER
    => I would expect SONOFF to go off after the saved Pulsetime is over

**My settings...**
#define SAVE_DATA 1
#define SAVE_STATE 0
#define APP_PULSETIME 150
#define APP_POWERON_STATE 0
MQTT
mqtt:broker.qos=2
mqtt:broker.async=true
mqtt:broker.retain=true

When I cut power, SONOFF is first off and comes then ON again after about 7 seconds. It seems to pull info from mosquitto and turns then on forever and ignores pulsetime

lobocobra, 2017-07-22:
By reading through all issues, I found out that MQTT server (mosquitto) retains some power states, which overrides the SONOFF status. But this still does not explain why PULSETIME is not respected.
=> how can I force SONOFF to enforce pulsetime?

It is not a problem if the motors are running 15 seconds, but if they would run several days until I find it out, they would burn for sure. As well I will have to have one SONOFF for UP and another for DOWN
=> I must be sure, that a SONOFF is OFF and I do not make a short circuit

davidelang, 2017-07-22:
The problem is that if the mqtt server is retaining the setting, then every time 
the device turns itself off (at the end of the pulsetime), the mqtt server will 
turn it on again. It's been told to make sure the device remains at the state 
it last saw the command for.

you need to disable retention in mqtt and your control software.

to detect this problem, look at the logs on your device (you may need to 
increase the logging level), you woudl see the commands arriving from the mqtt 
server telling it to turn on.

you should also look at adding limit switches to your motor that disable 
movement in that direction when at the limit.

lobocobra, 2017-07-22:
Thanks for the hints.
=> I already disabled rentention and it improved the situation, but randomly I still get the problem
=> I will dig deeper into the cause by increasing the log levels on the MQTT server side, maybe I have an issue with openhab as well, asking for retention of the status.

When I know more I will update the topic here.

lobocobra, 2017-07-22:
I increased log level to the maximum and still get to the point that PULSETIME is NOT respected, when SONOFF was ON on the time of a power-cut.
=> I could fix the retained issue, but the PULSETIME issue looks to me more like a CODE issue. 
=> PULSETIME has the value "on", if the device suffers a power-cut during it is on, I would expect the complied value or the last value

I will try to search in the code the place, where pulsetime is set and set it to the compliling value, if MQTT wants to set it to "on", which is an invalid value.

lobocobra, 2017-07-22:
I found a brute force hack, that solves my problem.
=> replace in the tab SONOFF follwoing code...
original: uint16_t pulse_timer[MAX_PULSETIMERS] = { 0 }; // Power off timer
NEW   : uint16_t pulse_timer[MAX_PULSETIMERS] = { 1 }; // Power off timer
=> this activates the timer at the beginning and as Pulsetime is at "0" during boot, it turns it off after 1 second

The solution would be to load the PulseTime earlier in boot sequence and then it would work more nicely.... I hope I find in the code the place where to load it.

lobocobra, 2017-07-23:
I adapted the code a bit:
**user.config.h:** 
#define APP_PULSETIME_BOOT     APP_PULSETIME     // [if PulseTime] is > 0 you want the deive to turn off automatically, but after a reboot it can remain off all time, this hack prevents it

**"sonoff" Tab** 
**replace:**
  uint16_t pulse_timer[MAX_PULSETIMERS] = { 0 }; // prevent after an power failure that the device go on 
**with:**
#ifdef APP_PULSETIME_BOOT
  uint16_t pulse_timer[MAX_PULSETIMERS] = { 1 }; // prevent after an power failure that the device go on a ON status forever, when PulseTime is compiled in
#else
  uint16_t pulse_timer[MAX_PULSETIMERS] = { 0 }; // turn PulseTime off on boot, when value of Pulsetime is at 0
#endif  // APP_PULSETIME_BOOT

lobocobra, 2017-07-24:
As I found a work-around I close the topic. Personally I believe it could be fixed much more professional, but it seems that it is only affecting a small population... the ones that want to use it as door opener or for curtains.

closed 2017-07-24 by lobocobra