arendst/Tasmota #629: different action with same config

jormus, 2017-07-23:
hi,
I have a strange thing going on with my sonoff touch..
I flashed both the latest version, and set button topic.
sonoff 1 sends toggle after 1 tap, and switch relay when double tap
sonoff 2 sends toggle by double tap, and switch state by one tap.

Cant figure out how that happend :-(

For now its ok, I need one sonoff that is inverse (toggle relay by multi tap), but how can I find the revers setting?

ionciubotaru, 2017-07-23:
read this [https://github.com/arendst/Sonoff-Tasmota/issues/200#issuecomment-306617207](https://github.com/arendst/Sonoff-Tasmota/issues/200#issuecomment-306617207)

jormus, 2017-07-24:
thx, setoption11 is the guilty one :-)

closed 2017-07-24 by jormus