arendst/Tasmota #630: Offtopic: connecting multiple sonoffs to the same AP

ionciubotaru, 2017-07-23:
I have a raspberry pi 3 configured as AP using (hostapd and isc-dhcp-server) + mosquitto server. I tried to connect around 90 sonoffs, unfortunately only 32 can
connect at the same time. The DHCP pool is large enough. Hostapd maximum clients is 200 (default value). 

I solved the problem using 3 raspberry pi.
Does anybody know where this limitation coming from or found a better solution ?

strasharo, 2017-07-23:
What's the wireless adapter and which driver is it using?

davidelang, 2017-07-23:
anything in the logs on the pi? enable debug logging if needed.

ionciubotaru, 2017-07-23:
It's a pi3 with on board wireless adapter and the latest raspbian Jessie lite version:

```
Version:July 2017
Release date:2017-07-05
Kernel version:4.9
```

arendst, 2017-07-23:
I suppose you do not have a parameter for hostapd max_num_sta set to 32?

ionciubotaru, 2017-07-23:
hostapd.conf:

```
interface=wlan0
driver=nl80211
hw_mode=b
channel=11
wmm_enabled=0
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

ssid=test
wpa_passphrase=test12345

```

I tried with 
```
hw_mode=g
```
and the same

arendst, 2017-07-23:
What happens if you add max_num_sta = 64

ionciubotaru, 2017-07-23:
I tried with 200 and nothing changed. From hostapd documentation:

```
# Maximum number of stations allowed in station table. New stations will be
# rejected after the station table is full. IEEE 802.11 has a limit of 2007
# different association IDs, so this number should not be larger than that.
# (default: 2007)
max_num_sta=255
```

strasharo, 2017-07-23:
It depends on the driver used for the wireless adapter:
https://www.raspberrypi.org/forums/viewtopic.php?f=36&t=76604

ionciubotaru, 2017-07-23:
`driver=nl80211` (standard driver provided by raspbian for Pi3)
According to article could be hardware or software limitation. If it is hardware we have no choice, but if it is a software limit do you know how to change it ?

strasharo, 2017-07-23:
nl80211 is the new 802.11 netlink interface public header and not the driver for your wireless adapter. Unfortunately I don't have access to a PI3, however IIRC it has a broadcom wireless, so the driver should be something like brcmfmac. Research the options to tune it.

ionciubotaru, 2017-08-31:
After few months of testing I found that raspberry pi 3 has some problems handling 32 concurent wifi connections (lag, wifi blocked). I replaced raspberry pi3 with [Zyxel](http://www.zyxel.com/support/SupportLandingSR.shtml?c=gb&l=en&kbid=MD09005&md=NBG-419N) 1 unit costs less than 10 EUR and handle 31 concurent connections like a charm.

closed 2017-08-31 by ionciubotaru