arendst/Tasmota #631: fulltopic stat promt - 5.3.0

timtimsson, 2017-07-23:
Hi,
i change from 5.1.7->5.3.0
If i toggle POWER with WebIF - stat is ok, but if i toggle with mqtt fulltopic is wrong?
5.1.7 works perfect....

```
 -mosquitto_pub -d -t /ESP_09/cmnd/POWER -m "OFF"
/ESP_09/cmnd/POWER OFF
stat/ESP_09/RESULT {"POWER":"OFF"}
stat/ESP_09/POWER OFF

-toggle with WebIF - ok
/ESP_09/stat/RESULT {"POWER":"ON"}
/ESP_09/stat/POWER ON
/ESP_09/stat/RESULT {"POWER":"OFF"}
/ESP_09/stat/POWER OFF

CMND: fulltopic
MQTT: /ESP_09/stat/RESULT = {"FullTopic":"/%topic%/%prefix%"}
```

Thanks

arendst, 2017-07-24:
I guess you named the mqttclient name and topic name the same resulting in equal fallback topic and topic. Making mqttclient different from topic should solve your problem.

See file releasenotes.ino regarding recent fallback topic change.

timtimsson, 2017-07-24:
... ohoh - thanks, it works with diffrent names-You're right!

closed 2017-07-24 by timtimsson