arendst/Tasmota #632: Sonoff 4CH - GPIO2

noxx2015, 2017-07-23:
hello

i flashed my Sonoff 4CH (Rev2.0) with Tasmota. I think the I02 is the
GPIO2 Port. How can i use this eg. for my temperature sensor.

i the config i use Sonoff 4CH and have only 4 feelds the setup the Sonoff (Relay 1-4).

lobocobra, 2017-07-24:
I have a similar question. I would like to USE on GPIO2 by attaching a switch to it.
=> when the switch is on (high), I would like to get an MQTT answer or directly act on POWER

Has anybody a hint, how to do this... I found nothing so far in Google, Wiki

davidelang, 2017-07-24:
define GPIO2 as a switch, and then set switchmode for the switch you define 
(i.e. switchmode1 for switch1) and set switch topic to have it send a mqtt 
message
k

noxx2015, 2017-07-24:
i change in the config menu the GPIO2 to my DS18x20 sensor. in the user_config.h i add #define USE_DS18x20. 
Red on VCC, Black on GND and Yellow cable an I02. But i get no values.

Program version	5.2.4
Build Date & Time	2017-07-24T18:26:34
Core/SDK version	2_3_0/1.5.3(aec24ac9)
Uptime	0 Hours
Flash write count	76 at F9000
Boot count	23
Reset reason	Software/System restart
Friendly name 1	Sonoff#CH4#1.1
Friendly name 2	Sonoff#CH4#1.2
Friendly name 3	Sonoff#CH4#1.3
Friendly name 4	Sonoff#CH4#1.4
 
AP2 SSId (RSSI)	myhome2_AP (20%)
Hostname	sonoff-2924
IP address	192.168.1.71
Gateway	192.168.1.1
Subnet mask	255.255.255.0
DNS server	192.168.1.1
MAC address	60:01:94:86:2B:6C
 
MQTT	Disabled
 
Emulation	Hue Bridge
mDNS Discovery	Enabled
mDNS Advertise	Webserver
 
ESP Chip id	8792940
Flash Chip id	1327185
Flash size	1024kB
Program flash size	1024kB
Program size	475kB
Free program space	528kB
Free memory	28kB

Amli2011, 2017-08-14:
Do you have a pull up resistor from the data line on the sensor? According to the data sheet it needs a pull up resistor. 4.7kohm. Without it you will not get any data from ad ds18x20. The rest looks right to me.

You might also set the telemetry time (I don't remember exact command, but it's in the documentation.)to a value other than zero to get it to automatically send temp at the interval you prefer.

I did set it one time via the console

closed 2018-05-07 by stale[bot]