arendst/Tasmota #633: Diasable MQTT Password

noxx2015, 2017-07-23:
how can i disable user and password in mqtt?

lobocobra, 2017-07-23:
I use mosquitto and I just leave in the mqtt.cfg file the 2 lines for user PW commented:
#<broker>.user=<user>
#<broker>.pwd=<password>
You can of course also delete the 2 lines.

But the question belongs rather to a MQTT area and not here. In the user_config.h you leave the 2 lines below as they are:
  #define MQTT_USER            "DVES_USER"       // [MqttUser] Optional user
  #define MQTT_PASS            "DVES_PASS"       // [MqttPassword] Optional password

stefanbode, 2017-08-05:
on mosquitto configuration file there is an option for anonymous allowed. If this is set to TRUE you can come with any password and any user and even empty ones and have full access.

https://mosquitto.org/man/mosquitto-conf-5.html

```
allow_anonymous [ true | false ]
Boolean value that determines whether clients that connect without providing a username are allowed to connect. If set to false then another means of connection should be created to control authenticated client access. Defaults to true.

Reloaded on reload signal.
```

closed 2018-05-07 by stale[bot]