arendst/Tasmota #634: TH10 / AM2301

cputoaster, 2017-07-23:
I just flashed a TH10 with 5.3.0 and got an AM2301 sensor (from sonoff) connected to it. I configured it via the Web page (am3201 on gpio14), can get working switch functionality and state messages, but nothing on sensor input. According to https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/issues/39 , there can be a problem with the contacts. I did try with and without the case, several times (with plugging in / out the case), and I still have the same problem, never ever saw anything appearing (eg with Status 10). Is there a way to measure if the sensor is wired correctly and the plug is connecting as it should?

arendst, 2017-07-26:
Enable more logging with ``weblog 3`` or ``seriallog 3``. You should at least see if the sensor is found.

cputoaster, 2017-07-26:
Tried that, I see nothing at all apart from

`12:57:27 DHT: Timeout waiting for start signal low pulse`

I guess the question is how I should test the wiring / sensor. Is there someone with a working sensor that can tell me the what resistance should be on the pins of the plug, so I can check which part is broken?

davidelang, 2017-07-26:
On Wed, 26 Jul 2017, Andres Meyer wrote:

> I guess the question is how I should test the wiring / sensor. Is there someone with a working sensor that can tell me the what resistance should be on the pins of the plug, so I can check which part is broken?

the communication with the sensor if digital, it's not a matter of resistance. 
The itead page for the TH10 shows the pinout of the sensor plug, there is 3.3v, 
signal, ground and an extra pin. you would need a logic analyzer or scope to 
watch the signal pin, not a simple meter.

cputoaster, 2017-07-26:
Thanks for the quick answer. Yes, I just thought that there could be some way of checking if the plug on the sensor cable is wired correctly without taking it apart. It does not seem to connect at all, so I guess a logic analyzer would not show anything atm.

cputoaster, 2017-07-26:
So, I opened the sensor, detached the plug in the sensor, measured the voltages and found that Gnd (red??) / Out (green) had been wrongly attached. I swapped them and everything works fine.

closed 2017-07-26 by cputoaster

piggz, 2019-02-03:
@cputoaster What should the plug be wired as? Im also getting null values, have (from left to right):
Red - (no description)
Green - Ground
White - Out
Black - VCC

How should these relate the the 3.5mm jack connector?

piggz, 2019-02-03:
Managed to fix it ... the wiring was correct, but one of the pins on the Sonoff side of the the jack, the one nearest the top of the connector was bent and not making a connection.