arendst/Tasmota #636: Random power off

0n3-70uch, 2017-07-24:
Hello,

I still notice some random power off. I dont know why and its totally random with different devices. I tested different Sonoff Versions and different Domoticz versions... no solution found. Hope someone can help me here.

Sonoff Logfile after a random power off:
`12:20:13 MQTT: stat/Sonoff_E89D96/RESULT = {"Webserver":"Active for ADMIN on Sonoff-7574 with IP address 192.168.2.101"}
12:20:14 HTTP: Webserver active on Sonoff-7574.local with IP address 192.168.2.101
12:20:26 mDNS: Initialized
12:20:26 MQTT: Attempting connection...
12:20:27 mDNS: Query done with 0 mqtt services found
12:20:27 MQTT: Connected
12:20:27 MQTT: tele/Sonoff/LWT = Online (retained)
12:20:27 MQTT: cmnd/Sonoff/POWER = 
12:20:27 MQTT: domoticz/in = {"idx":12,"nvalue":0,"svalue":""}
12:20:27 MQTT: stat/Sonoff/RESULT = {"POWER":"OFF"}
12:20:27 MQTT: stat/Sonoff/POWER = OFF`

And the same from Domoticz (one hour difference):
`2017-07-24 12:48:40.388 MQTT: Topic: domoticz/in, Message: {"idx":25,"nvalue":0,"svalue":"30.4;70.2;1"}
2017-07-24 12:53:40.615 MQTT: Topic: domoticz/in, Message: {"idx":25,"nvalue":0,"svalue":"29.8;69.9;1"}
2017-07-24 12:58:40.470 MQTT: Topic: domoticz/in, Message: {"idx":25,"nvalue":0,"svalue":"29.0;71.0;3"}
2017-07-24 13:03:40.694 MQTT: Topic: domoticz/in, Message: {"idx":25,"nvalue":0,"svalue":"28.4;71.7;3"}
2017-07-24 13:07:31.988 MQTT: Topic: domoticz/in, Message: {"idx":5,"nvalue":0,"svalue":""}
2017-07-24 13:08:41.198 MQTT: Topic: domoticz/in, Message: {"idx":25,"nvalue":0,"svalue":"27.6;72.4;3"}
2017-07-24 13:13:41.420 MQTT: Topic: domoticz/in, Message: {"idx":25,"nvalue":0,"svalue":"27.0;73.8;3"}
2017-07-24 13:14:58.873 Incoming connection from: 66.249.93.51
2017-07-24 13:14:58.985 Incoming connection from: 66.249.93.49
2017-07-24 13:18:12.801 MQTT: Topic: domoticz/in, Message: {"idx":11,"nvalue":0,"svalue":""}
2017-07-24 13:18:26.555 Incoming connection from: 192.168.2.20
2017-07-24 13:18:41.378 MQTT: Topic: domoticz/in, Message: {"idx":25,"nvalue":0,"svalue":"26.5;74.5;3"}
2017-07-24 13:20:27.990 MQTT: Topic: domoticz/in, Message: {"idx":12,"nvalue":0,"svalue":""}`

Greetings
Fabi

davidelang, 2017-07-24:
increase your logging level, and send the log to a remote syslog server, this 
may catch the logs from just before it shuts down.

0n3-70uch, 2017-07-24:
Oh good idea... thanks.

0n3-70uch, 2017-07-24:
I got a log... but don't know if its normal:
`Jul 24 16:39:27 192.168.2.131 Sonoff-3501 ESP-Wifi: Checking connection...
Jul 24 16:39:27 192.168.2.131 Sonoff-3501 ESP-Wifi: Connected`

ionciubotaru, 2017-07-25:
It's normal, the wifi library checks periodically if the wifi connection is still alive, then prints the result `Connected`. This messages are displayed only in the DEBUG LEVEL.

0n3-70uch, 2017-07-25:
Hi, I found out some interesting things:

1) only some of the Sonoffs power of randomly (3 of 19 total)
2) these devices working very slow (a very slow response also the web server is slow). Sometimes there is absolutely no response...
3) how to get them back to life: reset via button help for a short time. But only a reset command (Reset 2) helps. 

Ok this sounds really a bit mysterious. Seems that's something is wrong with the flash?

Greetings
Fabi

ionciubotaru, 2017-07-25:
It looks like a power supply problem. 
I programmed till now 90 sonoffs: 
- 2 of them (sonoff POW) has power supply problem, one doesn't send 3V3 at all, the other power off like yours after 3-5 muinutes
- 3 of them (sonoff Touch) - 2 has power supply problem, one has esp8266 problem (write it, but crash at boot)

0n3-70uch, 2017-07-25:
Oh that sounds bad. Did you find the error in the power supply? The 3,3V Step down or something else?
As I can see the ESP8266 is working with only 2,5V...

davidelang, 2017-07-25:
I would check wifi signal levels in those areas (try swapping with one that 
works) before deciding it as bad flash.

a poor power supply on the board is more likely
k

0n3-70uch, 2017-07-25:
Yes I also changed the Sonoff with another one. It's definitive a problem with the Sonoff. A closer connection to the Wifi hotspot speed up the Sonoff a bit but not really. Only the full reset helps at the moment. The Wifi strength is round about 58%.

ionciubotaru, 2017-07-25:
sonoffs works fine with mqtt even at 8% wifi (I have few in the house in this situation).
of course you cannot update them OTA, but for daily usage is OK, it doesn't resets for weeks

I din't check the hardware parts of bad sonoffs yet.
Try to supply 3V3 via FTDI and check if it is stable (don't use mains and ftdi in the same time).

0n3-70uch, 2017-07-25:
Thanks the bypass via FTDI is a good idea... I will update the status here after the test.

arendst, 2017-07-25:
Do they really power off and stay off or do they just restart? If they restart perhaps changing the way wificonfig is configured might solve it. Try ``wificonfig 4`` to disable restarts on wifi problems.

0n3-70uch, 2017-07-25:
It seems that they just go off... and then no response. I already tested your trick with `wificonfig 4`... not working. I will make the FTDI power test this evening.

0n3-70uch, 2017-07-26:
Hi, need help with the logging again:

`Jul 26 10:15:29 192.168.2.131 Sonoff-3501 ESP-MQTT: tele/Sonoff/STATE = {"Time":"2017-07-26T09:15:29", "Uptime":11, "Vcc":3.167, "POWER":"ON", "Wifi":{"AP":1, "SSID":"Briese", "RSSI":24, "APMac":"34:81:C4:B5:D5:8D"}}
Jul 26 10:15:39 192.168.2.131 Sonoff-3501 ESP-RSLT: Receive topic domoticz/out, data size 298, data {
   "Battery" : 255,
   "RSSI" : 12,
   "dtype" : "Temp + Humidity",
   "id" : "82025",
   "idx" : 25,
   "name" : "Außen > Sensor > Weg",
   "nvalue" : 0,
   "stype" : "THGN122/123, THGN132, THGR122/228/238/268",
   "svalue1" : "22.5",
   "svalue2" : "81.8",
   "svalue3" : "3",
   "unit" : 1
}`

This was captured with the device inside the house. Why is there a logging of the sensor outside the house? The interesting fact: These two devices make trouble... Do they influence each other?

Greetings
Fabi

arendst, 2017-07-26:
As you enabled logging AND configured domoticz for this unit you see that Domoticz is swamping MQTT with all kinds of (useless) information. If you log it longer you'll see all your sensors and switches and come along...

This is not your power problem.

You will have to filter the logging data for the correct device yourself.

0n3-70uch, 2017-07-26:
Ah thanks. So Domoticz is spaming into the network. I'll wait a bit and check the logfiles later. There was a power off around one hour ago. But I stoped the logging yesterday... Now I need to wait, write you later :-).

davidelang, 2017-07-26:
make sure they have different names and different IP addresses

the devices don't know anything about each other, if they are sending confusing 
messages, it's because they've been configured in a way to conflict

0n3-70uch, 2017-07-26:
Oh both devices working with different names and IP adresses. It's more a Domoticz "problem" like Theo told me...

arendst, 2017-07-26:
It's not a Domoticz problem. The Domoticz spamming does not cause your power problems.

It's caused by something different...

0n3-70uch, 2017-07-26:
Yeah I understand. We have two situations:
1) Messages from other devices. Happend because of Domoticz.
2) The power off problem... Therefore I will check the logfile after a crash.

EmilySmith0, 2017-07-27:
When you flash esp with arduino, it does not erase its full memory. Few eeprom registers contains values from previous firmwares which are causing power-off problem.(I may be wrong but given solution works)
Try to erase full memory with link given below (1M file) and reinstall tasmotta.
Once I was having problem of "power off" within 1-2 seconds and erasing memory did the trick

Emily

EmilySmith0, 2017-07-28:
If you update firmware on the air(OTA), this issue may occur.
Its a good practice to erase full memory before firmware installation or upgrade.
However no one  can tell what the issue was until perform debugging.

Emily

0n3-70uch, 2017-07-29:
I build a script for upgrade process. First the upgrade and later the `reset 2` with flash erase. After that procedure I reconfigure the Sonoffs. I will try the next upgrade with `reset 2` before... Thanks for help

Greetings Fabi

0n3-70uch, 2017-08-12:
Hi, there was a new reboot today. I got only this information:
> Software/System restart
The signal quality is 44%.

Greetings Fabi

0n3-70uch, 2017-08-14:
I will close this issue and open a new question "how to reduce power offs after sytsem restart":
https://github.com/arendst/Sonoff-Tasmota/issues/733

Thanks for help...

closed 2017-08-14 by 0n3-70uch