arendst/Tasmota #637: 4 dallas sensors - MQTT String corrupt

Schnittcher, 2017-07-24:
Hello,

on the Sonoff Switch, we have 4 dalls sensors.
MQTT sends a corrupted string:

`{"Time":"2017-07-24T12:23:35", "DS18x20":{"DS1":{"Type":"DS18S20", "Address":"10D4B431030800E6", "Temperature":19.0}, "DS2":{"Type":"DS18S20", "Address":"105A7F30030800C2", "Temperature":19.8}, "DS3":{"Type":"DS18S20", "Address":"10BB863003080089", "Temperature":19.2}, "DS4":{"Type":"DS18S20", "Address":"10C72D3203080028", "Temperature":19.8}}, "TempUnit":"`

When we have 2 sensors on the switch, the MQTT string is fine:

`{"Time":"2017-07-24T12:24:52", "DS18x20":{"DS1":{"Type":"DS18S20", "Address":"10D4B431030800E6", "Temperature":19.1}, "DS2":{"Type":"DS18S20", "Address":"10C72D3203080028", "Temperature":19.8}}, "TempUnit":"C"}`

Can anyone help?
Sorry for my ba english.

davidelang, 2017-07-24:
it looks like you are hitting the max message size limit, increase it.
kkkk

Schnittcher, 2017-07-24:
Thanks, I will try it.
The max message size is MESSZ, right?
Can I change this with a command? Or I need to flash it, again?

davidelang, 2017-07-24:
you will need to flash it. You may need to increase the max size in the library 
(the instructions tell you to increase it to 256 IIRC)
k

arendst, 2017-07-26:
You could also use the pre-compiled binary ``sonoff-ds18x20.bin`` which allows for more sensors.

In the next version I'll update the default buffer length anyway with some 4 bytes as the problem is introduced by adding the TempUnit field to the JSON message.

closed 2017-11-07 by arendst