arendst/Tasmota #639: Sonoff basic: flashing works fine (no error), but no serial output or any other sign of life

tr1plus, 2017-07-24:
Hello

I got my sonoff today and tried it out. The default firmware worked fine so I know the sonoff is (was) working.
Next step was to flash tasmota firmware. I got the latest version and followed all the steps. Flashing went fine. I reassembled everything, connected it to power and... nothing.
I spent a while looking trough the issues: #611 seems to be relevant but the solution is either not very clear or doesn't work for me.

What I've tried:
- Setting the flash mode to dout
- switching tx and rx on my adapter
- flashing with the button held in
- holding the button for 4 seconds when connected to the adapter

Nothing seems to be working. I don't think I broke my sonoff (yet) as I'm still able to flash.

Any solution/insight in this problem?

Thanks and kind regards
Joren

ionciubotaru, 2017-07-25:
1.check the arduino serial window parameters in the right lower corner
2.use putty to open you serial port at speed 115200

tr1plus, 2017-07-25:
baud rate is the same as the upload speed
I'm not home right now, but would putty show anything else as the arduinoIDE serial console? My Sonoff doesn't even blink the LED or anything so I'm thinking there is more to it?

I spent yesterday night reading and it seems maybe a new sonoff version was released which doesn't work with custom firmware?

ionciubotaru, 2017-07-25:
All sonoffs are based on esp8266 so Tasmota will work as fine as the original firmware.
Try this: 
- connect serial ftdi
- press gpio0
- remove and reconnect power to sonoff (keep ftdi connected to computer, remove power line between ftdi and sonoff) - keep gpio0 pressed during this process

If you see some garbage on the serial (try putty) is ok.
If not, you have a problem with serial, ftdi, etc.

tr1plus, 2017-07-25:
just for my information. To get serial output, do I need to keep tx/rx connected and connect the main power. Or only power in via the FTDI? I read that it should only be FTDI right and I should never connect mains and ftdi at the same time?

I will try flashing an older verion of tasmota tonight. I read in another issue that was the solution for someone. Maybe the new sonoff basic is not compatible after all: #608, #463 (read more down in the issue)

Thanks for helping already

ionciubotaru, 2017-07-25:
power sonoff via ftdi during debugging or flushing
do not connect mains and ftdi in the same time

ionciubotaru, 2017-07-25:
I received 10 sonoffs basic and few sonoffs touch last week from banggood.com - is cheaper  than itead :) -and I was able to write all of them. Entering boot mode by pressing gpio0 at startup is hardware encoded in esp8266 so itead cannot change this behavoiur

tr1plus, 2017-07-25:
I also got mine from banggood. I'll test again tonight but I have followed all those procedures.

What version did you use to flash?

Thanks

ionciubotaru, 2017-07-25:
arduino 1.8.1
esp8266 for [arduino v2.3..0 ](https://github.com/esp8266/Arduino)
Tasmota 5.3.0

tr1plus, 2017-07-25:
Ok, i'll verify the versions tonight.
One additional question, I also heard not all FTDI adapters are good. I got the following one from banggood: https://www.banggood.com/FT232RL-FTDI-USB-To-TTL-Serial-Converter-Adapter-Module-For-Arduino-p-917226.html?rmmds=myorder

Would this one be good or should I get a more trustworthy one?

ionciubotaru, 2017-07-25:
I use [FOCA](https://www.itead.cc/foca.html)  for more than 5 years and works perfect
try to decrease the upload speed, maybe 115200 is too much for your ftdi chip

tr1plus, 2017-07-25:
Ok, so I followed your instructions step by step (using same versions as you) and I got weird characters in console... I guess that's a good start.

Connecting the sonoff to main power again does nothing. No LED, button does nothing... I honestly have no clue anymore :(
http://imgur.com/a/9TLFK <-- output of console + serial gibrish

Thanks for helping, but I'm at a loss :(

tr1plus, 2017-07-25:
Ok, I also just realised I was using 5v, my FTDI connector has a weird thing you have to swap from one side to the other.
I'm not sure if I damaged anything. I still got serial output so I guess it is not a real problem?

tr1plus, 2017-07-25:
tested with some other firmware (https://github.com/KmanOz/Sonoff-HomeAssistant)
Same results; I get weird characters in serial console, but once I plug it in no response at all.
Seems I am only able to flash when I put the connector on 5v, 3.3v doesn't even allow me to flash.

Is my sonoff just broken?

davidelang, 2017-07-25:
feeding the esp8266 5v _will_ fry it, so if you've tried to boot it at 5v, I 
expect it's dead.
k

tr1plus, 2017-07-25:
but I still get serial gibrish, so that tells me it (may be) is alive? or am I mistaken there?

davidelang, 2017-07-25:
once you've sent it 5v, all bets are off, you may be getting just noise, or 
there may be some parts that work and it's only partially toasted, but you 
really should just throw it away and try with a different one.

5v has been shown to very reliably fry these chips.

tr1plus, 2017-07-25:
ok, guess it's time to order a different one. Let's hope for better luck next time then :-)

LavaGlass, 2017-07-26:
I had the exact same problem as you and solved it here. https://github.com/arendst/Sonoff-Tasmota/issues/611

tr1plus, 2017-07-26:
Hey

I saw your issue earlier too. Your solution was

> Figured it out and now all my Sonoff's are working! Just had to choose the right settings in the Arduino IDE. Make sure you have it set to 1M No SPIFFS.

As far as I know I am using the correct settings everywhere as mentioned on the wiki. I even tried playing around with them but no luck :(

Thanks
Joren

Makr91, 2017-07-28:
So, I tried flashing with the ESP8265 Chipset instead of the 8266 and voila it worked! 

I  even had my TFDI adapter set to 5v, at one point, and I had flashed a nodemcu prior and forgot to change the settings over and thought I ruined it. but, flashing with the ESP8265 chipset at 1M 64K seems to have worked, OTA and all appear to be working for me

tr1plus, 2017-07-28:
@Makr91 that may be good news for me. Could you share all your arduinoIDE flash settings please, so I can replicate those.

How recent (and from what seller) is your sonoff if I may ask.

Thanks!

arendst, 2017-07-28:
If the chipset is causing you this problem I would suggest to read the Wiki > Upload section.

As of version 5.3 I only support Flash Mode DOUT (which is default for esp8285) as this mode works fine for all type of esp8266 devices.

Your suggested 1M 64k spiffs won't work over time. You definitely need 1M 0k spiffs as documented on that page too. See page Wiki > Prerequisite how to install this 1M 0k spiffs linker file if it is not available or in PlatformIO use the provided platformio.ini files.

Makr91, 2017-07-28:
Sonoff Vendor: https://www.amazon.com/gp/product/B06X6NBDFW/ref=oh_aui_detailpage_o04_s00?ie=UTF8&psc=1

First two were ESP8266, second 4 were ESP8265

I did the prerequisite instructions to a T and could select the 1M no SPIFFS with the ESP8266 chipset, it would flash but nothing would happen(no serial, no lights).  I will work on getting the setting for 1M 0K spiffs into my IDE settings for the ESP8285 chipset.

Yes, I did switch it back and got it to flash

https://dallas.m4kr.net/uploads/2017-07-28_03-42-46-bb08c8d2-8484-4b16-ac8e-9180c3c904a7.png

To be clear, I first flashed this with nodemcu settings on 5v, then attempted reflash with correct settings but used the 1M 64K as some of the documentation still shows this. 

Then I randomly tried the ESP8265 Chipset with the 1M 64k setting select, the rest were as close to the original settings as possible.

tr1plus, 2017-07-28:
hmm okay. I will try one more time, but I remember having no luck flashing on 3.3v (it didn't allow me to upload).

I'll try again :) Thanks for the info

davidelang, 2017-07-29:
On Fri, 28 Jul 2017, Mark wrote:

> To be clear, I first flashed this with nodemcu settings on 5v

As soon as you apply 5v to these devices, all bets are off, you may very well 
have fried part of the system.

Makr91, 2017-07-29:
I concur, however after review this page, I think my sonoff were the older model:

https://github.com/arendst/Sonoff-Tasmota/wiki/Upload

Arduino IDE

Make sure you configured the IDE as described in Prerequisite!

Load the file sonoff.ino into the IDE.

In the Arduino IDE for sonoff select from Tools Board Generic ESP8266 Module ( Tools Board Generic ESP8285 Module for CH4 version) and set the following options:

Upload Using: Serial
Flash Mode: DOUT
Flash Frequency: 40MHz
CPU Frequency: 80MHz
Flash Size: 1M (64K SPIFFS) ⚠️️**If Version 5.x.x -> Flash Size: "1M (no SPIFFS)"**⚠️️
Debug Port: Disabled
Debug Level: None
Reset Method: ck
Upload Speed: 115200
Port: Your COM port connected to sonoff

Makr91, 2017-07-29:
> As of version 5.3 I only support Flash Mode DOUT (which is default for esp8285) as this mode works fine for all type of esp8266 devices.

Pretty sure that if you are following Bruhs guides, that he using a different setting, i am pretty sure this is key to the flash.

declan2511, 2017-08-01:
@Makr91 Your solution works,  changing the board flash mode to DOUT works every time.  It seems ITead have changed the spec.  I have just received five new Sonoff Basic units.  All have a sticker on the board with YD QCPASS printed.  All boards failed to flash initially, until I changed the flash mode.  I now have flashed 5.0.0 directly and OTA upgrades to 5.5.0 with no apparent problems.  

Initially WiFI settings seemed to be a problem.  After flashing the device, it kept defaulting to the indebuurt2 AP,  tried everything to set it.  I had configured user_config.h.  Eventually, I created a guest network and configured it to my internal network through the web console.  I will update the post if I can figure out why this was happening  - it could be related to my router which has issues!   

BTW great work @arendst on the software

Thanks

tr1plus, 2017-08-08:
Got a new one and followed the instructions. Seem to be working now. Have a different problem but I open a new issue. This one is resolved!

Thanks for the support!

closed 2017-08-08 by tr1plus

gregkeys, 2017-09-05:
Got 5 new sonoff basics, I can flash fine but they show no sign of life after..... has something changed with this latest batch?

Makr91, 2017-09-05:
Did you  change the board flash mode to DOUT?

gregkeys, 2017-09-05:
yeah i even tried flashing just blink in the Arduino IDE with DOUT
selected. it succeeds in flashing but there is no blink

Makr91, 2017-09-05:
So when it is flashing, the lights flash, but nothing happens afterwards?

can you screenshot your IDE setup prior to flashing?

gregkeys, 2017-09-08:
I've tried both IDE's Arduino and Platform, I've tried swapping the rx/tx around and I've tried the esp8265 pretty much everything I can think of, I can get it to sucessfully upload but there is just no life, I wonder if its not how i've wired up the arduino nano.

[image]
[image]
[image]

ozgurgesli, 2017-09-08:
I'd just like to confirm that, my newly arrived sonoff basic, one of the ones with the fix/patch cap at the solder side, was flashing without problems, but no activity afterwards (version v.5.6.1) when i was flashing through Arduino IDE with DIO flash mode, but then, after i changed to DOUT, it flashed and started working.

declan2511, 2017-09-08:
Hi Greg, I have had similar intermittent problems.  Software reports a successful flash but the device doesn't respond.  The device also doesn't produce any messages through the serial interface.  In these cases I have had to reset the device by flashing other firmware, typically ESPeasy, ESPpixel stick or by flashing a blank image through NodeMCU. Once you have a responsive device re-flash with this firmware.  No idea what causes this, but it seems specific to the newer sonoffs

funnyfeet, 2017-10-07:
I had the same error earlier today. No SSID, No LED and no response to the manual switch. I'd tried flashing with both esptool.exe and NodeMCU flasher. I had 2 different USB-TTL adapter a "sochange" and a TA-02L. Both appeared to work fine but left my Sonoff dead. In desperation I tried a 3rd adapter I was lucky enough to find at the bottom of the bits bin - a LoveElectronic barebons FTDI. Without changes it worked first time. So in my case, the issue was clearly the adapter.
This might be related but I've not yet checked specifically....https://hackaday.com/2014/10/22/watch-that-windows-update-ftdi-drivers-are-killing-fake-chips/

gregkeys, 2017-10-07:
> hackaday.com/2014/10/22/watch-that-windows-update-ftdi-
> drivers-are-killing-fake-chips/

@funnyfeet excellent information.

perhaps we should start a wiki with all the devices that work, the ones
that don't and the ones which need special setups and configurations

agengr1293, 2017-11-26:
So 1M no spiffs is not an option in my ESP8266 settings.  Are there instructions somewhere to get this option?  I am running Arduino 1.8.5 and esp8266 ver 2.3.0

agengr1293, 2017-11-26:
Man I have tried all options and am unable to flash the Sonoff board.  Any suggestions here.  It keeps giving me esp_comm_upload_mem failed errors.  I followed the prerequisite and everything is good in Arduino, but just doesn't communicate.

declan2511, 2017-11-26:
Hi agengr1293, which sonoff device are you trying to flash. That message usually pops up  when the device is not in flash mode.  Depending on the device it can be tricky to get into flash mode, usually the green light doesn't come.  Make sure you have the toggle switch fully clicked down when you power on the sonoff.

agengr1293, 2017-11-27:
I bought the ITEAD WiFi Wireless Smart Switch Module ABS Shell Socket.  Does that make a difference?  I will go thru you suggestion this evening, because I was waiting for the light to come on first prior to hitting the flash mode button.

agengr1293, 2017-11-27:
OK here is the process that I really need help on since I am still getting espcomm_upload_mem failed errors.  Do I need to change anything?
1) Unplug USB connecting to FTDI and Sonoff.  (FTDI and Sonoff are connected via jumpers).
2) Press flash button and hold on Sonoff
3) Plug in USB to PC
4) Let go of flash button on Sonoff
5) Upload from Arduino

When I tried this there was no flash on the Sonoff, but still getting errors on communication.

Any help would be greatly appreciated.

pelson, 2018-01-13:
I was flashing a Sonoff RF Bridge with my own firmware but wasn't seeing it actually getting loaded. After quite some effort, I stripped the firmware back to a basic hello world and ended up adding ```board_flash_mode = DOUT``` to my platform.ini configuration. I also held the "reset" button for the entire programming process. Finally, I saw success! 🎉 
Thanks for the explicit reminder about the DOUT flashmode. 😄

TLCary, 2018-02-28:
Sonoff Dual R2, same thing.  I took a close look at the chip and it's an 8285.  Things change.

JerryLoz, 2018-03-06:
Agree with TLCary - Newest Sonoff basic's from Banggood use an 8285 *not* an 8266. Changed CPU in Arduino to 8285 and flashed first time every time. 
Prior to change, esptool.py failed using "auto" for CPU detection. ESP8266Flasher.exe failed as well. I guess now we know why...
From all of the postings, it is clear more than one problem is occurring - so everybody has to keep their eyes open, I think.

shamsherr, 2018-05-28:
i think i figured it out, i checked with a voltmeter and even thought the FTDI VCC is reading 3.4 volts the rx and tx pins are reading 4.7v and 5v respectively, i think this could be wreaking havoc with the esp8266, can anyone confirm this?