arendst/Tasmota #640: how to add pir sensor

yzg011, 2017-07-25:
how to add pir sensor

davidelang, 2017-07-25:
wire it to a GPIO and define it as a switch (make sure it's 3.3v compatible or 
use a resister divider network to make it 3.3v safe)

cobra85ec, 2017-07-26:
what model PIR sensor can be used with sonoff basic..?

yzg011, 2017-07-26:
HC-SR501

yzg011, 2017-07-26:
yes i chose 09 switch  but always on and off

davidelang, 2017-07-26:
On Wed, 26 Jul 2017, cobra85ec wrote:

> what model PIR sensor can be used with sonoff basic..?

anything that has as it's output a relay or a signal that you can convert to 
3.3v vs ground for the two states (i.e. just about anything)

kutteg, 2017-08-02:
I have added a HC-SR501 on GPIO 4 successfully on a sonoff t16 , but I don't want the sensor to toggle the reley, I want it only to publish MQTT messages for the state changes as other sensors (eg. AM2301). How do I have to configure that?

Thanks

davidelang, 2017-08-02:
look at switchtopic
k

kutteg, 2017-08-02:
tried that already, it always triggers the relay

can you give me further help?

davidelang, 2017-08-02:
what settings did you use?
k

kutteg, 2017-08-02:
switchtopic 1

davidelang, 2017-08-02:
you need to actually set a different topic to use.

qtronik, 2017-08-23:
So... by curiosity ... what's the solution!? It's my next project too!

jhlweb, 2017-10-26:
You have to use Switchmode<X> 1 and SwitchTopic "XXXX" for this.
See my post in #420

petonic, 2017-12-13:
Qtronik, specifically, look [at this link](https://github.com/arendst/Sonoff-Tasmota/wiki/Expanding-Sonoffs#connect-switch).  If you only have one relay (the default config for a Sonoff basic) and if you don't have a separate SwitchTopic, then Switch1 (on GPIO 14, presumably) will have direct control of the relay.

As soon as you configure a SwitchTopic, then the Sonoff will MQTT send that topic, and will  do nothing with the relay.  Took a second pass through reading it before I noticed that, myself.

cobra85ec, 2018-01-31:
i did that but not i keep getting On and OFF every 3 sec...

joann235, 2018-02-28:
Same here. Cycles on and off every few seconds. Completely stumped.

SodoffBaldrick, 2018-02-28:
Did you use Switch1 or Switch2? If 1, then try selecting Switch2 and try again

stephen-coates, 2018-06-01:
I junked my SR-501s and replaced them with SR-505s, 3.3v quarter of the size and a lot less false triggers. There's no sensitivity or delay adjustment but out of the box the coverage seems similar to the 501 and they seem a lot more stable. The only downside is I now have to make some nice collars for them to fill in the 501 sized holes in my upvc soffits.

demian85, 2018-06-18:
Please help. I'm trying to use an SR-501 with Wemos D1 mini, but after trying many configurations, nothing seems to make sense. I keep getting hunderds of messages like
```
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
{"Time":"2018-06-18T10:24:32","Switch1":"ON","AM2301":{"Temperature":20.0,"Humidity":55.8},"TempUnit":"C"}
```

for the SENSOR topic. 

Look at the time, why is this happening? It also never goes OFF!

closed 2018-08-17 by stale[bot]

qtronik, 2021-04-16:
> If someone is still having this issue, FYI: I fixed it connecting the hc-sr501 to 5V. I also found a german wiki page which says that one can shortcircuit the diode near the 5V pin to get everything running with 3.3V, because internally the boards runs 3.3V and has Voltage reduction on the input to be used with 5V.
> 
> https://wiki.fhem.de/wiki/Sonoff#Spannungsversorgung_des_HC-SR501

Perfect! It really work for me for stability! I've removed de diode and short pads with solder.
My PIR is connected to an SonOff-mini 3v3 and the diode WAS dropping it at 2.45v...
Success and no more "light party" in my kitchen ;-)