arendst/Tasmota #641: Hue Emulation - No Discovery in Alexa

pprotschka, 2017-07-25:
hi everybody,
i enabled Hue emulation on a Sonoff 4CH and trying to get discovered by Alexa but it is not showing up. 
Is there a "press hue bridge button" command in tasmota or is this enabled all the time?

I have also tried physical power cycling and software reboots but still no go.

i have other devices working fine and the switch is responding normally via mqtt. What would be a good way to debug this? where do i start?

pprotschka, 2017-07-25:
addon: 
i have made sure devices are named uniquely in the config.

arendst, 2017-07-26:
Enable more logging with command ``weblog 3`` and monitor the console output. Expect something like:
```
09:55:03 Wifi: Checking connection...
09:55:03 Wifi: Connected
09:55:15 UPnP: HUE 3 response packets sent to 192.168.2.211:1901
09:55:15 HTTP: Hue API (/null)
09:55:16 HTTP: Hue API (/null)
09:55:22 HTTP: Hue API (/config)
09:55:23 Wifi: Checking connection...
09:55:23 Wifi: Connected
09:55:25 UPnP: HUE 3 response packets sent to 192.168.2.53:4583
09:55:25 UPnP: HUE 3 response packets sent to 192.168.2.53:4583
09:55:25 HTTP: Hue API (/null/lights/1/state)
09:55:25 HTTP: Hue POST args ({"on":true,"bri":19})
09:55:25 RSLT: RESULT = {"POWER1":"ON"}
09:55:25 RSLT: POWER1 = ON
```

pprotschka, 2017-07-26:
thanks for the reply.

Is there a **"just wait"** component to this? i have checked again this morning and tada the devices show up as expected. Yesterday they would not show up for about 30 minutes. 

. from the log:
09:10:10 UPnP: HUE 3 response packets sent to 1.1.1.9:4353
09:10:10 UPnP: HUE 3 response packets sent to 1.1.1.9:4353
09:10:11 UPnP: HUE 3 response packets sent to 1.1.1.10:60806
09:10:11 UPnP: HUE 3 response packets sent to 1.1.1.10:60806
09:10:12 UPnP: HUE 3 response packets sent to 1.1.1.9:44864
09:10:12 UPnP: HUE 3 response packets sent to 1.1.1.10:44872
09:10:17 UPnP: HUE 3 response packets sent to 1.1.1.102:50000
09:10:17 UPnP: HUE 3 response packets sent to 1.1.1.102:50000
09:10:22 UPnP: HUE 3 response packets sent to 1.1.1.9:44864
09:10:22 UPnP: HUE 3 response packets sent to 1.1.1.10:44872
09:10:26 MQTT: tele/sonoff/STATE = {"Time":"2017-07-26T09:10:26", "Uptime":0, "Vcc":3.177, "POWER1":"OFF", "POWER2":"OFF", "POWER3":"ON", "POWER4":"OFF", "Wifi":{"AP":1, "SSID":"XXXX", "RSSI":100, "APMac":"XXX"}}
09:10:30 UPnP: HUE 3 response packets sent to 1.1.1.10:53072
09:10:30 UPnP: HUE 3 response packets sent to 1.1.1.10:53072
09:10:32 UPnP: HUE 3 response packets sent to 1.1.1.9:44864
09:10:32 UPnP: HUE 3 response packets sent to 1.1.1.10:44872
09:10:33 UPnP: HUE 3 response packets sent to 1.1.1.10:2673
09:10:33 UPnP: HUE 3 response packets sent to 1.1.1.10:2673
09:10:42 UPnP: HUE 3 response packets sent to 1.1.1.9:44864
09:10:42 UPnP: HUE 3 response packets sent to 1.1.1.10:44872

closed 2017-07-26 by pprotschka

m3zm3riz3, 2017-08-27:
Did you manage to fix this as i have the same issue?

vks007, 2019-07-18:
I also faced these issues with setting up the Hue. I think you have to time the discovery. Tasmota sends a Hue advertise signal every 1 min or so. You can see when it sends it in the console logs by changing the weblog level to 3.  Make sure your alexa is trying to discover devices when Tasmota sends the Hue broadcast message.