arendst/Tasmota #642: MQTT, send a TOGGLE cmnd only if the buttonTopig also differs from the groupTopic

qorron, 2017-07-25:
since I now have got a round tuit [0], I'll write the issue as promised[1]:
If the buttonTopic is set and differs from the switchTopic a TOGGLE command is sent instead of the ON/OFF command depending on the current state of the module's own relay because it is assumed that the state of the switche that gets the command is not known to the module and thus it is impossible to choose from either ON or OFF.

if modules are grouped together via the groupTopic this can lead to inconsistencies once a message gets lost. i.e. one switch is out of line with the others.

to remedy this, the buttonTopic could also be compared to the module's groupTopic. the module then is aware of the current state of the topic by looking at its own state.

inconsistencies my still happen or be even intentionally created[2] but are resolved the instant one of the grouped switches is activated.

[0] http://666kb.com/i/dl8s114f80tsz84lo.jpg
[1] https://github.com/arendst/Sonoff-Tasmota/issues/553#issuecomment-314156933
[2] Example: ambiance and high powered lights are controlled both with a set of wall switches, turning both on or off like legacy light switches. if a moody light scene is desired, the ambiance light can be switched one independently via an other device e.g. phone. (yes, I'm planing to build such an installation)

arendst, 2017-07-26:
Implemented in next release.

jormus, 2017-07-28:
you could do that with node-red. I can give you a example function that saves the state of the relay in a global variable, that you can use in all you're flows. Gives you the option to do something depending on the state of the relay. (and off course the option to add more controllers). I personal don't use group-topics, and prefer the use of node red, gives me an visual of what is going on :-)

qorron, 2017-08-01:
thanks, that helps a lot!!

btw: git will serve your needs much better if you would commit more often.
e.g. use at least one commit per solved problem/implemented feature.
changes are much more readable that way. you can use tags to designate version numbers and branches for features that are more complex to build.

that's a good template to model ones own strategy:
http://nvie.com/posts/a-successful-git-branching-model/
some background:
http://www.creativebloq.com/web-design/choose-right-git-branching-strategy-121518344

some more resources, the ultimate git cheat sheet:
http://ndpsoftware.com/git-cheatsheet.html
and this helps if you think you messed something up:
http://justinhileman.info/article/git-pretty/git-pretty.png

all these resources have made my life so much easier.

arendst, 2017-08-09:
@qorron just found your answer back in the pile! I knew I had read it but couldn't find it again.

I'm looking into commits and have started making a development branch as the main one. Will look into your suplied links soon.

closed 2017-08-09 by arendst