arendst/Tasmota #643: Can a WS2812 be controlled via MQTT?

mihalski, 2017-07-26:
I have the WS2812 shield for Wemos D1 and have been trying to figure out if it's possible to control the state of the light (on/off, colour, blinking) via MQTT but have not being able to find any way to do so.

Is this something that Tasmota is capable of?

Regards,
Michal

arendst, 2017-07-26:
Mine is working fine. So keep digging.

[image]

[image]

mihalski, 2017-07-26:
Where is that 2nd image from? I've never seen it.
This is my config:
[image]
I only have 1 pixel in this config.

arendst, 2017-07-26:
The second image is my own unsupported in-house front-end running on an apache server.

That config works fine with me. See Wiki commands specific for WS2812.

mihalski, 2017-07-26:
Seems I had to break a bit of a barrier of understanding to get it working.

But I'm still unclear about how I would make it blink at a certain rate. Sending POWER ON/OFF commands repeatedly seems like a hack.

Would I need to modify the code to add my own "scheme"?

joecotton, 2017-07-26:
See the Wiki:
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#main

You can set the BlinkCount and BlinkTime, and instead of sending POWER
ON/OFF, send POWER 3 or POWER BLINK to start blinking, and POWER 4 or POWER
BLINKOFF to stop (prior to BlinkCount being reached.)

mihalski, 2017-07-26:
Oh!! I had NO idea this applied to other lights connected! I thought it only applied to the relay linked light.

Thank you! It probably would have taken me weeks to stumble upon that.

closed 2017-07-26 by mihalski