arendst/Tasmota #646: Sonoff POW bad topics (anyone seen this?)

tascar01, 2017-07-26:
Yesterday I installed two sonoff basic's and a sonoffpow, all serial and at the same time using Arduino IO, set-up each with the browser access.

The two basic's work OK, but the POW (being set to POW) shows data in the browser, but the MQTT topics are as below.

I can see the topics in mqtt-spy, but can't pickup the topics from Node-red or Home Assistant.

[image]

Anybody seen this? or has a solution. I used 5.3.0 code

arendst, 2017-07-26:
Check your prefix1, prefix2, prefix3 and fulltopic settings via the web console.

tascar01, 2017-07-26:
Sorry not sure where you mean, only topics I could find on the web console (192.168.1.19) is in the Configure MQTT Tab, this is the list from the Information Tab.

 
[image]
or am I looking in the wrong place?

arendst, 2017-07-26:
- Goto the console webpage
- Type prefix1 and hit return key
- Read respone on console
- Same for prefix2 and prefix3.

If strange results see the wiki comma ds section and fill with cmns, stat and tele.

tascar01, 2017-07-26:
I did all three and they returned 
[image]
but with 2 and 3 for the others, what do I meed to change?

arendst, 2017-07-26:
What is a console....

One of the sonoff webpages is called Console. Just select the console butten and start entering commands please.

tascar01, 2017-07-26:
Sorry I didn't understand, the output is
[image]

arendst, 2017-07-26:
And compare with one of your other sonoffs. Then just enter that info via the console: prefix1 cmnd, prefix2 stat and prefix3 tele.

tascar01, 2017-07-26:
This is from one of the basic done at the same time and is working OK
[image]

arendst, 2017-07-26:
So what's the next step? Hint read my last entry.

tascar01, 2017-07-26:
If strange results see the wiki commands section and fill with cmnd, stat and tele.

Will have a good look and let you know if I sort it, or don't.

Many Thanks, will leave this open until a result

tascar01, 2017-07-26:
Many Thanks, all sorted and working.

closed 2017-07-27 by arendst