arendst/Tasmota #647: Switch relay on temperature setpoint

jvanderkroon, 2017-07-26:
Can you add a temperature setpoint to the webintface. When the setpoint is reached the relay must be switched.

with the current firmware i use node-red to read the mqtt sensor data and when the setpoint is reached i send a mqtt message to swith the relay. I think it's a nice feature to do this standalone on the Sonoff.

davidelang, 2017-07-26:
no, the current firmware does not take local actions on the temps. You are the 
second person today to ask for this feature :-)

pprotschka, 2017-07-27:
third. ;)
im messing around with esp easy just because this is missing. Tasmota is so great but there needs to be an "offline failsafe". Servers go down, nodered get hung up etc etc...

closed 2018-05-07 by stale[bot]

vch911, 2018-06-15:
I'd like to support this request. Local simple rules would be awesome. In my case it's humidity. I want to keep a fan in a bathroom on until humidity is back to normal. Anyway, Tasmota software already has this kind of hard rules: switch or pushbutton triggers relay directly. Why temperature/humidity cannot?