arendst/Tasmota #648: [Request] Support for Magic Home Wifi for LED Lights (ESP8266)

ChrSch84, 2017-07-27:
Hello i have a request, is it possible to flash Tasmota at the Magic Home Wifi for LED Lights from Aliexpress (or others).

It has a ESP8266 Chip maybe it is possible. I like Tasmota @my sonoff devices :) So maybe on other devices too.

Thanks

[Magic Home Wifi](https://www.aliexpress.com/item/16Million-colors-Wifi-RGB-RGBW-led-controller-smartphone-control-music-and-timer-mode-magic-home-wifi/32755262961.html)

ionciubotaru, 2017-07-28:
define a custom device in sonoff_template.h

```
......................
const mytmplt modules[MAXMODULE] PROGMEM = {
  { "Sonoff Basic",    // Sonoff Basic (ESP8266)
     GPIO_KEY1,        // GPIO00 Button
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     0,                // GPIO02 
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_USER,        // GPIO04 Optional sensor
     0,                // GPIO05
     0,                // GPIO06 (SD_CLK   Flash)
     0,                // GPIO07 (SD_DATA0 Flash QIO/DIO/DOUT)
     0,                // GPIO08 (SD_DATA1 Flash QIO/DIO/DOUT)
     0,                // GPIO09 (SD_DATA2 Flash QIO)
     0,                // GPIO10 (SD_DATA3 Flash QIO)
     0,                // GPIO11 (SD_CMD   Flash)
     GPIO_REL1,        // GPIO12 Red Led and Relay (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Green Led (0 = On, 1 = Off)
     GPIO_USER,        // GPIO14 Optional sensor
     0,                // GPIO15
     0,                // GPIO16
     0                 // ADC0 Analog input
  },
......................
```

arendst, 2017-10-30:
Implemented in master release v5.9.0

closed 2017-10-30 by arendst

gi1mic, 2017-11-01:
I installed the latest Tasmota from source on my magichome. I understand this has only been implemented and it seems to work via the command line interface but:

a) Can or how do I control brightness and colour from Domoticz/mqtt
b) Does the web interface only support brightness or am I missing something.

Btw, Thanks for the implementing this!

arendst, 2017-11-02:
Domoticz cannot control RGB (See #981) but it can control brightness if you configure a switch as a dimmer.

The web interface is minimal to conserve code space and therefore does not provide color control.

HobbytronicsPK, 2019-03-24:
Use RGB WW Virtual Switch in Domoticz for Colour / Warm / Cold / Brightness etc. features