arendst/Tasmota #649: h801 all leds flash on when applying power

jormus, 2017-07-27:
hi,

I take away the voltage of the h801 controller when i'm not using it, but when I apply voltage, the leds flash on for like 50 ms. This looks like somebody is flashing with a camera, because my led strips are RGBWA all in one. Can this be disabled somehow?

thx

arendst, 2017-07-28:
I guess not as it's the result of initial power up of the esp8266 having set the pwm ports to a default stae before the firmware kicks in.

Same happens on my BN-SZ ceiling led. Just leave the power on and turn the led off.

closed 2018-05-07 by stale[bot]

ramafun, 2018-09-13:
I have the same issue. I don't think it is due to a HW problem, because when I leave the original firmware and configure with ewelink, it doesn't blink on powering up.

andriej, 2019-01-22:
Exactly the same issue

bernhardhauffen, 2019-06-09:
I have the same issue. With the orignial FM the leds are not flashing while powering up.