arendst/Tasmota #651: socket error on client <unknown>

snizzleorg, 2017-07-27:
I can't get my S20 to connect to mqtt.
The console shows the following:

```
19:56:58 mDNS: Query done with 0 mqtt services found
19:56:58 MQTT: Connect FAILED to 192.168.178.94:1883, rc 5. Retry in 10 seconds
```

The server is mosquitto with now user name and password, which I also configured on the S20.

```
Jul 27 18:55:58 automator mosquitto[25391]: 1501181758: New connection from 192.168.178.124 on port 1883.
Jul 27 18:55:58 automator mosquitto[25391]: 1501181758: Socket error on client <unknown>, disconnecting.
```

even though I have set the client ID to the following:

```
19:57:45 CMND: MqttClient
19:57:45 RSLT: outlet1/RESULT = {"MqttClient":"sonoff-4631"}
```

Any idea what might be wrong here?

ionciubotaru, 2017-07-28:
First try a public mosquitto server like test.mosquitto.org or cloundmqtt.com
Second check your mosquitto version and update it to the latest stable release.

snizzleorg, 2017-07-28:
Thanks! I got it working. I had to take out the mosquito password file from the configuration. So somehow mosquito was expecting a password even though other clients worked fine without....

closed 2017-07-28 by snizzleorg

AalianKhan, 2019-01-03:
@snizzleorg I am getting the same error, so which file do you have to take out exactly?