arendst/Tasmota #652: Help, friends on how to set the button set!

1043717432, 2017-07-28:
2 channel = on, 1 channel auto = off and 3 channel auto = off, 3 channels = on, 1 channel = 1 1 and 2 channels automatically = off, 4 channels = on and off = off 1/2/3 / channel, can this be implemented?

davidelang, 2017-07-28:
please try to re-state what you are trying to do, it's not clear what you are 
asking.

1043717432, 2017-07-28:
@davidelang   Hello there!
   Is this, I think the firmware used in the garage door and aisle switch, my switch is self-resetting jog switch, scene 1: when I open the switch 1, and then open the switch 2, this time switch 1 will automatically shut down.
Scenario 2: The switch 1 is turned on and the switch 2 is automatically turned off. If the switch 3 is turned on, the switch 1 and the switch 2 are automatically turned off. Scenario 3: When I turn on the switch 4, switch 1, switch 2, switch 3, if it is in the state, will become off.

davidelang, 2017-07-28:
the tasmota firmware does not do this sort of thing in the firmware, you would 
do this sort of thing in the home automation software.

The tasmota firmware is focused on providing the ability to control the switch 
to automation software.

arendst, 2017-07-28:
Using the Sonoff 4ch pro in interlock mode does just that; the microcontroller only allows one relay active or no relay active.

Installing Tasmota on this device allows also mqtt control providing the same interlock mode.

1043717432, 2017-07-28:
How to use mqtt to complete this operation?tks

1043717432, 2017-07-31:
Hello, the author.
I was esp8266 development board, can the firmware choose Sonoff 4ch pro?

davidelang, 2017-07-31:
It's not clear what you are asking, but Tasmota can be installed on any esp8266 
and you select the module type that matches the hardware that's attached to the 
esp.

1043717432, 2017-07-31:
I would like to ask how to use 8266esp + relay, for example: Sonoff 4ch pro interlock mode is the case, in the firmware web how to choose sonoff 4ch pro firmware?

davidelang, 2017-07-31:
this project supplies the tasmota firmware (and it's source code), not the 
sonoff 4ch pro firmware, you would have to contact sonoff to get details on 
their firmware.

As I understand it from watching the discussions, there may be additional 
hardware involved with some of those features.

the plain sonoff setting will let you have an esp8266 + relay and will work just 
fine. With only one relay, "interlock mode" is a meaningless term.

If you have multiple relays, Tasmota doesn't have any controls that care about 
the state of one when changing the state of others, that is up to your home 
automation software. Tasmota just provides a way for that automation software to 
communicate with the esp8266 and control the relay(s) that are connected to it.

1043717432, 2017-07-31:
Thank you! Is my ability to express well! This diy is not getting it for the time being. @davidelang

1043717432, 2017-07-31:
@davidelang  Thank you! Is my ability to express my mistakes! This diy is not getting it for the time being.

closed 2018-05-07 by stale[bot]