arendst/Tasmota #655: Sonoff 4CH Rev 2.0 - GPIO Bug ?

noxx2015, 2017-07-28:
hello 

i try 4 versions of Tasmota (also sonoff-ds18x20.bin e.g. Rev 5.4.0) and ESPeasy, on 2 different Sonoffs 4 CH (both Rev 2.0).
Also 4 different DS18b20. But non of them show me values of my temp-sensor :-(

is a hardware bug in rev 2.0 possible?

`

Program version | 5.4.0
-- | --
2017-07-25T16:38:37
2_3_0/1.5.3(aec24ac9)
22 Hours
144 at F7000
58
Software/System restart
Sonoff1
Sonoff2
Sonoff3
Sonoff4
 
FRITZ!Box Fon WLAN 7390 (90%)
sonoff-4707
192.168.1.72
192.168.1.1
255.255.255.0
192.168.1.1
60:01:94:85:72:63
 
192.168.1.9
1883
DVES_857263
iobroker
sonoff
sonoffs
cmnd/sonoff/
 
Hue Bridge
Enabled
Webserver
 
8745571
1327185
1024kB
1024kB
474kB
528kB
24kB

`

[image]

[image]

arendst, 2017-07-28:
Where do you connect the ds18b20 sensor?

noxx2015, 2017-07-28:
DS18B20:
Black on GND
Red on VCC33
Yellow on I02

[image]

arendst, 2017-07-29:
I guess you will need a 10k resistor between 3V3 and IO2

noxx2015, 2017-07-29:
thx, thats was it

[image]

closed 2017-07-29 by arendst