arendst/Tasmota #656: SmartThings device handler

BrettSheleski, 2017-07-28:
I'm not sure where else to post this, but others may find this useful.

I wrote a SmartThings device handler to support Sonoff devices running the Sonoff-Tasmota firmware.  It implements the Momentary, Switch, Refresh, and Polling capabilities by making HTTP requests to the Sonoff device from the SmartThings hub.  Currently you can turn on, turn off, and toggle the output of the Sonoff as well as read the current state of the Sonoff (on or off).

For those already invested in the SmartThings ecosystem, you can use the device handler to control the Sonoff devices from the SmartThings app.

The device handler can be found here:
https://github.com/BrettSheleski/SmartThingsPublic/blob/master/devicetypes/BrettSheleski/sonoff-tasmota.src/sonoff-tasmota.groovy

## Instructions
### Add Device Handler
1. Log in to the SmartThings IDE (https://graph.api.smartthings.com/)
2. Go to `My Device Handlers`
3. Click `Create New Device Handler`
4. In the `From Code` tab paste in the code from https://github.com/BrettSheleski/SmartThingsPublic/blob/master/devicetypes/BrettSheleski/sonoff-tasmota.src/sonoff-tasmota.groovy
5. Click `Create`
6. Click `Publish` --> `For Me`

### Add Device
1. Log in to the SmartThings IDE (https://graph.api.smartthings.com/)
2. Go to `My Devices`
3. Click `New Device`
4. Give it a Name, And Device Network Id
5. For Type scroll to the bottom of the dropdown and select 'Sonoff-Tasmota'
6. Click `Create`
7. In the table that displays the different properties of the newly created device, click the `edit` link next to Preferences.
8.  Enter the IP Address of your Sonoff and specify Port 80 (and username and password if applicable).
9.  Click `Save`

Now the device is added to your account and you should be able to control your device from the SmartThings app on your phone.

kiranmk702, 2017-07-29:
@BrettSheleski, How do i know if switch is running Sonoff-tasmota, is there any url from where you bought them from pls ?

arendst, 2017-08-09:
Thnx. Did add it to the Wiki.

closed 2017-08-09 by arendst

BrettSheleski, 2017-08-09:
I ended up copying and pasting the instructions to a readme.md file in my GitHub repo.

Since you're just adding a link in the wiki to this issue, it may be better to have the link point to my readme.md file instead.

https://github.com/BrettSheleski/SmartThingsPublic/blob/master/devicetypes/BrettSheleski/sonoff-tasmota.src/readme.md

I noticed someone has forked the repo already and looks to be adding a reboot option. If they submit a pull request to me then I can update the readme.md file with additional info.

arendst, 2017-08-09:
Thnx again. Just updated the wiki to your readme.md

djtonyd, 2017-09-17:
The device handlers works well and creating a new device works, but when turning the sonoff on/off the little indicator ON symbol on the app always stays in the ON/Green mode and not off, in saying this the actuall sonoff does switch on and off as per pushing the button on the app, but  when pushing the app button it doesnt change state to off, it just keeps staying on the the whole time even when the relay does come in and out. any ideas?

BrettSheleski, 2017-09-18:
I think I know what you're trying to say, but I'm not 100% certain.  Let me first try to restate the issue so we're on the same page.

You are able to turn on/off the Sonoff as you would like using the SmartThings app, however the status it is reporting is not updating appropriately.  Therefore when you turn the Sonoff On, it still shows that it is Off after successfully turning On.

Here's some background:

When turning the Sonoff On/Off with the SmartThings app it is actually issuing the Toggle command to the Sonoff.  Therefore it should correctly turn it from On to Off, or from Off to On regardless of what the current status SmartThings reports.  It does this by issuing an HTTP command to Sonoff and attempts to parse the response JSON (within the Parse() method of the SmartThings device handler).  This happens sometime after issuing the toggle command to the Sonoff.  The JSON that is returned by the Sonoff includes the current state which gets should get parsed and update the status of the Sonoff reported by the SmartThings app.

The Device Handler also implements the Polling capability (see the Poll() method in the device handler).  The SmartThings hub will call the Poll() method periodically (exactly when and how often is unknown) which send another HTTP command to the Sonoff and that response again gets parsed by the Parse() method.  This is done to update the status of the Sonoff in SmartThings app in case it's status has been set by some other means.  The Parse() method reads the response of the HTTP command and updates the status of the Sonoff indicated by the SmartThings app.

It sounds like the Parse() method is not behaving correctly.  However I just did some testing while writing this and it appears to be working correctly for me.  If you press the refresh tile/button in the SmartThings app is the status updated appropraitely?

Do you happen to have a PulseTime set on your Sonoff?  I have a Sonoff 1-Channel relay attached to my garage door opener set with a PulseTime of 200 milliseconds and its status is sometimes reported incorrectly by the SmartThings app.

djtonyd, 2017-09-18:
Yes it it now working it took awhile for the time delay, but it is now showing correctly when sonoff is on and off,

While in talking to you, would you know of any good wifi dimmer devices that the sonoff has or any out there in the market that supports the smartthings? I'm in need of a wifi dimmer for my lounge room to dim my incandescent lights,

Regards Anthony

BrettSheleski, 2017-09-19:
I got a bunch of GE Z-wave smart dimmers.

https://www.amazon.com/dp/B01MUCZA1C/ref=cm_sw_r_cp_apa_cggWzbT94W0R9

They're Z-wave, not WiFi, but SmartThings works great with them. They're about $40 each though.

djtonyd, 2017-09-19:
Yer I looked at them I don't really want to replace the wall switch with that, I'm thinking a push button and the device hides behind the wall, I was looking at these

https://aeotec.com/z-wave-light-dimmer-switch

Have you had them or know anyone that has? I live in Australia and I'm pretty sure the z-wave frequency is illegal here lol, that's why I was hoping to get a wifi device dimmer?

https://www.amazon.com/dp/B01MUCZA1C/ref=cm_sw_r_cp_apa_cggWzbT94W0R9

muknattapak, 2017-09-19:
I have got the same problem like @djtonyd it still ON symbol on the smartthings app and always stays in the ON/Green mode and not off but when pushing the app button it doesnt change state to off, it just keeps staying on the whole time. Any idea? 
And another question is it can control on/off via the app?

BrettSheleski, 2017-09-19:
It sounds the SmartThings app is not communicating at all to the Sonoff.  Make sure the IP address of the Sonoff is correct in the SmartThings app.

The device handler does implement the switch capability (see http://docs.smartthings.com/en/latest/capabilities-reference.html#switch) so there are distinct on() and off()  methods.  Therefore if you are using Alexa to control the Sonoff via SmartThings you can say "Turn On ..." or "Turn Off ..." and it will turn the device on/off accordingly (not just toggle).

However when using the SmartThings app it simply tells the Sonoff to toggle it's state when pressing the button.  Therefore it shouldn't matter if the SmartThings app is reporting the wrong current state.

The only time I've seen the app report the wrong status is if the Sonoff is toggled from outside the SmartThings app (manual toggle, from the Sonoff-Tasmota web interface, MQTT command, etc.).  The SmartThings app does periodically poll the status of the Sonoff therefore it should update the status when that happens.  Therefore it is possible it is reporting the wrong status within the SmartThings app until it decides to poll the device again.

Also, within the SmartThings app you can press the refresh button (bottom-left) to manually update the status of the Sonoff.

cvanandel, 2017-12-25:
Is it possible to use the device handler for a dual device like a Sonoff Touch 2 gang, where you have 1 IP address but 2 switches. 

By the way Brett, fantastic job, it works great with my single devices.

BrettSheleski, 2017-12-26:
It may be possible, I do not know since I don't have such a device.

I have a feeling it may currently operate one of the outputs, but not the other.

I plan on rewriting the device handler some day where one would add a single "logical" device to smart things which may spawn child devices depending on the module type. It could also read how the GPIO pins are configured and spawn child devices for those capabilities as well.

blokeyuk72, 2018-01-10:
Hi Brett... The device handler works great and what you have done here is awesome. I do have the same problem as mentioned previously though. I can turn on and off from smartthings, or we interface, but it always reports status as a green "on".

Any ideas? 
Thanks in advance
Blokey

blokeyuk72, 2018-01-10:
Even when off!

BrettSheleski, 2018-01-10:
Since Tasmota changed the format of the HTTP responses to be "true" json (see #1363 ), I have not yet updated the SmartThings device handler to accomodate this change yet.

However @diablodale has forked my repo and made a change to handle the updated format that Tasmota now responds with.  This change will be merged into my code base sometime soon to account for this change, and also be backwards compatible with "legacy" deployments of Tasmota devices.  See [here](https://github.com/diablodale/SmartThingsPublic/pull/1) for more detail on the issue.

blokeyuk72, 2018-01-10:
hi Brett

Thanks for fast response. Look forward to the edit!

Real Newbie question here (Rea Newbie I am!!). What do you mean by "forked" and "Changed Handle"?
Thanks again.

enchoss, 2018-01-11:
Hello, i flashed my sonoff pow to Sonoff-Tasmota firmware and everything works fine, but 
Smartthings switch always shows that sonoff is ON 
and in graph.api.smartthings.com/device for this sonoff :

Data | No data found for device
Current States | No states found

So can anybody help me to correct this problem?

enchoss, 2018-01-11:
OK! I test @diablodale vertion and the result is the same.

blokeyuk72, 2018-01-17:
hi brett

I see that @diablodale and yourself have done some more editing to the device handler over the last few days, but unfortunately I still cannot get an accurate status returned from the ST hub. 

I am using Tasmota firmware version 5.11.1 and this latest handler but it still returns as "on" when it is off and also with the refresh button. 

I hope that you can find a solution or guide me in the right direction as I really would like this direct connection; I know I haven't currently got the skills to code it!!

In the meantime, I have found a workaround that works for me, and may work for others....

My sonoffs are wired into my mains lights by room and work fine apart from the status issue; I have found the following will return a correct status, and is a bit of a temporary workaround:

1) Go to your Sonoff device on your local network to bring up the sonoff's Tasmota "homepage" by putting the IP address into your browser.
2) click on "Configuration"
3) click on "Configure Other"
4) Click on "Hue Bridge" emulation and "Save". The sonoff device will then restart.
5) Click on "Add Thing" on your Smartthings App on your mobile.
6) The Smartthings Hub will then find 2 new devices after a little wait of about 30 seconds.
   a) Hue Bridge ######
   b) Sonoff Device
7) As the Hue Bridge is just an emulator, and a "Dummy" parent device, I rename it for instance to "ZZ Kitchen Light Hue Bridge" in the "new Device Page" on the ST App  so that it appears at the bottom of the Home Page on the ST App and doesn't get in the way/clutter things up.
8) I rename the Sonoff Device to "Kitchen Lights" or whatever.
9) Save it
10) Now on the ST app, I have my "Kitchen Lights" showing their correct status. It shows up in the graph device list as a "Lan Hue Extended colour", but I'm sure this can be changed by user if required. Seeing as the only function that is required with a sonoff switch is to turn the device on and off I don't see that this will really matter.
11) Now, I if I ask Google Home to turn off the Kitchen Lights etc then it works great and reports correctly. The only drawback I can see to this work around is If I as Google Home to "Turn off all the lights" it turns off all my lights including the emulated Hue Bridge attached Sonoff's. 

Great for me, but if anyone had their TV connected this way, then this single phrase would also turn off their TV.

You can still use this workaround for any appliance and change the Icon and name in ST to a "TV" etc; but if you do and then ask Google/Alexa to turn off "all" your lights then your TV will turn off for instance.

It's a compromise maybe until perhaps the Device Handler works for me; not to detract from all of the hard work going on here as I know that I really would like this to work, but maybe something in this that helps with the device handlers coding??

Thanks
Blokeyuk72

BrettSheleski, 2018-01-17:
Do you happen to have multiple Sonoff devices configured using the same IP address?

There's a side effect of the device handler changing the Network Device ID of the device to the hexadecimal value of the configured IP address.  If you have multiple configured with the same IP address there would be a collision here and may explain why the status is not getting updated properly.

If using the simulator, make sure to uninstall the device from there too.

blokeyuk72, 2018-01-17:
Hi

2 devices at moment, both sonoff basic, both with individual ip's!

Thanks

protingas, 2018-01-21:
hello. i have very big problem with sonoff flashing. every time then i try flashing sonoff basic all times i get this problem:
C:\Windows\system32>cd\Users\Mantas\Desktop\New folder

C:\Users\Mantas\Desktop\New folder>esptool.exe -vv -cb 115200 -cp COM3 -ca 0x000
00 -bz 1M -bm dout -cf tasmota.bin
esptool v0.4.12 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
        setting baudrate from 115200 to 115200
        setting port from  to COM3
        setting address from 0x00000000 to 0x00000000
setting flash size from 512K to 1M
setting flash mode from qio to dout
        espcomm_upload_file
        espcomm_upload_mem
        setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
        flush start
        setting serial port timeouts to 1 ms
        setting serial port timeouts to 1000 ms
        flush complete
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        read 0, requested 1
trying to connect
        flush start
        setting serial port timeouts to 1 ms
        setting serial port timeouts to 1000 ms
        flush complete
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
        espcomm_send_command: receiving 2 bytes of data
Uploading 495808 bytes from tasmota.bin to flash at 0x00000000
        erasing flash
        size: 0790c0 address: 000000
        first_sector_index: 0
        total_sector_count: 122
        head_sector_count: 16
        adjusted_sector_count: 106
        erase_size: 06a000
        espcomm_send_command: sending command header
        espcomm_send_command: sending command payload
        setting serial port timeouts to 15000 ms
        setting serial port timeouts to 1000 ms
        espcomm_send_command: receiving 2 bytes of data
        writing flash
..........................................................................................................................      read 0, requested 1
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01
 0x04
closing bootloader
        flush start
        setting serial port timeouts to 1 ms
        setting serial port timeouts to 1000 ms
        flush complete
error: espcomm_upload_mem failed

andrewdenny, 2018-01-25:
Hello, I am having the same issue with the current status of the Sonoff switch not being reported correctly in the ST app. The app switches the Sonoff on and off but the app shows that the switch is on no matter what the true state of the Sonoff is. I have only one Sonoff configured so there are no conflicting ip addresses. When I flash the Sonoff with Eric Maycock's firmware, the ST app reports the status correctly. I want to use the Tasmota firmware so please help us since a few of us have the same problem. I have flashed 8 Sonoff basics with the NODEMCU flasher with no problems at all.

BrettSheleski, 2018-01-25:
Try opening the Simulator in the SmartThings online IDE and select your actual device.  After the device is installed/updated click the on/off/toggle buttons and see if there's any errors that appear in the log.

Also, I'll add a configurable refresh interval to the device handler that will update the status from the device.

andrewdenny, 2018-01-25:
Two questions. I see now that certain Tasmota files were updated yesterday to 5.11.1e. Has the sonoff.bin file been updated too?  Second question. Since I use the bin file for flashing, can I change the project name and topic name of each Sonoff that I flash by logging into the web interface after?

andrewdenny, 2018-01-25:
ok I reflashed again and went into the simulator mode. I started with the switch in the off position, hit refresh, the app still indicates "on" even though switch is off.  then hit the "on" button on the app and turn the switch on, hit the refresh button, app still reads "on". hit the  "on" button on the app again to turn the switch off again. Still no change of the status of the switch in the app. Below is a log printo out of what I did.

1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:22 AM: debug parse()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:22 AM: debug 0050
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:22 AM: debug IP address entered is 192.168.1.70 and the converted hex code is c0a80146
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:22 AM: debug sendCommand(Status:null)
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:22 AM: debug getStatus()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:22 AM: debug POLL
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:20 AM: debug parse()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:20 AM: debug 0050
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:20 AM: debug IP address entered is 192.168.1.70 and the converted hex code is c0a80146
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:20 AM: debug sendCommand(Power:Off)
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:20 AM: debug OFF
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:20 AM: debug parse()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:19 AM: debug parse()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:18 AM: debug 0050
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:18 AM: debug IP address entered is 192.168.1.70 and the converted hex code is c0a80146
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:18 AM: debug sendCommand(Status:null)
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:18 AM: debug getStatus()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:18 AM: debug POLL
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:18 AM: debug 0050
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:18 AM: debug IP address entered is 192.168.1.70 and the converted hex code is c0a80146
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:18 AM: debug sendCommand(Status:null)
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:18 AM: debug getStatus()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:18 AM: debug POLL
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:13 AM: debug parse()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:13 AM: debug 0050
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:13 AM: debug IP address entered is 192.168.1.70 and the converted hex code is c0a80146
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:13 AM: debug sendCommand(Power:On)
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:51:13 AM: debug ON
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:47 AM: debug parse()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:47 AM: debug 0050
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:47 AM: debug IP address entered is 192.168.1.70 and the converted hex code is c0a80146
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:47 AM: debug sendCommand(Status:null)
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:47 AM: debug getStatus()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:47 AM: debug REFRESH
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:46 AM: debug parse()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:45 AM: debug 0050
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:45 AM: debug IP address entered is 192.168.1.70 and the converted hex code is c0a80146
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:45 AM: debug sendCommand(Power:Toggle)
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:45 AM: debug TOGGLE
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:45 AM: debug PUSH
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:44 AM: debug parse()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:43 AM: debug 0050
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:43 AM: debug IP address entered is 192.168.1.70 and the converted hex code is c0a80146
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:43 AM: debug sendCommand(Status:null)
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:43 AM: debug getStatus()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:43 AM: debug REFRESH
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:41 AM: debug parse()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:41 AM: debug 0050
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:41 AM: debug IP address entered is 192.168.1.70 and the converted hex code is c0a80146
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:41 AM: debug sendCommand(Power:Toggle)
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:41 AM: debug TOGGLE
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:41 AM: debug PUSH
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:37 AM: debug parse()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:37 AM: debug 0050
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:37 AM: debug IP address entered is 192.168.1.70 and the converted hex code is c0a80146
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:37 AM: debug sendCommand(Status:null)
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:37 AM: debug getStatus()
1cabbb78-30bf-4178-86e3-64eb56e058ee 6:50:37 AM: debug REFRESH

andrewdenny, 2018-01-25:
I also pushed the toggle on and off and the refresh from the simulator with the same results

BrettSheleski, 2018-01-25:
It looks like it's hitting the parse() method, which is good.  This means that the device handler is getting the response from Tasmota which contains the JSON string of the current status of the device.

Actually, the latest code for the device handler, which was modified recently to support #1363 , does not have this debug line in it any longer.  

You're running old code for the device handler.

Update your device handler code to the latest.  https://github.com/BrettSheleski/SmartThingsPublic/blob/master/devicetypes/BrettSheleski/sonoff-tasmota.src/sonoff-tasmota.groovy

andrewdenny, 2018-01-25:
> BrettSheleski/SmartThingsPublic/blob/master/devicetypes/BrettSheleski/
> sonoff-tasmota.src/sonoff-tasmota.groovy

ok deleted device and device handler, replaced with the one you pointed me
to and created a new device .... the same thing is still happening....
could it be my bin file? What method do you use to flash your switches..
could you zip the folder you use and send it to me via www.wetrransfer.com.
. Thanks.  There is something different going on if your Tasmota flashing
and Smartthings integration works as it should for you and not for me.  I
will probably eventually move over to a Raspberry PI using Openhab but I
want to get it working here first.

protingas, 2018-01-25:
anyone helps my? I brick 2 sonnof devices. every time when I try flashing its stock in "writing flash" state. everytime firmware it's more it's less stops when writing flash. anyone can help?  

writing flash
 .......................................................................................................................... read 0, requested 1
 warning: espcomm_send_command: didn't receive command response
 warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
 warning: espcomm_send_command: wrong direction/command: 0x01 0x03, expected 0x01
 0x04
 closing bootloader
 flush start
 setting serial port timeouts to 1 ms
 setting serial port timeouts to 1000 ms
 flush complete
 error: espcomm_upload_mem failed

stringpark, 2018-01-28:
Hello, I am having a similar problem to the other commenters with the status of the sonoff basic as reported in the SmartThings App. I am using Tasmota 5.11.1 and the latest device handler code. My installed configuration has GPIO14 set to "09 Switch 1" to work with a standard on/off light switch. In my case, sending the toggle command via the app will correctly turn the sonoff on and off and also show the correct status in the app. However, the status in the app will not update when using the physical on/off switch to toggle the sonoff. Using the refresh button in the app will correct the status, but it will not update automatically. Any ideas? Thx!

andrewdenny, 2018-01-28:
Hello everyone.... It gets weird now. After reading stringpark's post , I
changed my GPIO 14 to 09 switch 1 and saved. Using the Smartthings app to
control my switch, I now get a status change at each tap and the switch
responds. The only problem is that if I control the switch from the black
button i get no status change within the app, The only way to get the
correct status in the app is by hitting the refresh button in the app, just
as stringpark has stated. I went one step further and returned my GPIO 14
setting to 00 NONE which was the setting that was set with the Tasmota
flash. Remember that before the app was not changing its status no matter
where you controlled the switch from. Well now it is still only showing a
status change when controlled from the app. if controlled from the switch
the status will only change if you hit the refresh button.

Brett could you not add a refresh command to occur after each switch
command received or sent from the app, the black button and any GPIO
connected switch?.I still do not understand why all of a sudden the status
actually started to work in the app after I changed the setting of  GPIO
14. Common sense tells me that returning the setting to the default flashed
setting should result in the app not changing its status again. I think we
are getting closer to a solution here. Also one more thing.  I have also
noted that the switch response gets slower and sometimes the swith does not
toggle at all when MQTT is enabled from the configuration other menu. When
you disable MQTT the switch toggles from the app no matter how many times
you tap the screen. *Is this because it has to wait for communication from
the MQTT service? I have not started to use it so that is why I disabled it
in the first place and noticed the difference in response time. I am trying
to get all of my devices working perfectly in Smartthings before I go the
Openhab / MQTT route.*

stringpark, 2018-01-28:
Just for extra information, figured I'd mention that when using the device handler simulator, it appears that "poll" and "refresh" both produce the same result of correcting the status in the app. I'm not really educated on how this stuff works, but would the device handler code need to instruct the hub to "poll" the device every so often to get it's current status (in case it was changed locally via GPIO)? Maybe it already does that, I don't know.

danieljafari, 2018-02-01:
same problem with the status of Sonoff being reported incorrectly as always on. Curiously, I have linked ST to google home, and when I ask google home, it correctly reports that the switch is off, all the while ST app shows a green "On" in front of it. Also, I have flashed Tasmota on some Lyasi wall switches (https://community.smartthings.com/t/lyasi-wifi-wall-switches-sonoff-plus-ewelink-cloud/97244) and added them with this device handler and have zero issues with status reporting in ST. Any ideas?

andyjenkinson, 2018-02-04:
I noticed that hitting the web interface tends to slow down the switch until eventually it hangs, I think it runs low on system resources. Mine works ok if I disable MQTT (which I don't use anyway). Better is to remove it from the firmware on compile I guess.

Also I have noticed that pinging the sonoff on its IP addrrss results in two replies indicating a conflict on the network, though no conflict exists. Anyone else seen this? Mine work regardless but could be a clue.

andrewdenny, 2018-02-04:
I will check that ping scenario.... I also noted earlier that having the
MQTT enabled slows the response of the switch.

ledfortr, 2018-02-06:
Hey Brett (and all),
I have a Sonoff Touch (US) and I am trying to flash it with Tasmota.  I am struggling with what needs to be done with the GPIO0 to get it into maintenance mode.  I see some pictures where folks have soldered a wire to GPIO0, stating that it need to be grounded.  Does that mean it needs to connect to the GND pin header I soldered on?  Any help would be appreciated.  I'm a bit of a newbie at this so please be specific.  Thanks!

enchoss, 2018-02-06:
I just get GND from the programmer and hold it on GPIO0 until pluging the
cabel in USB port, then i relece the wire.

ledfortr, 2018-02-06:
Perfect!  Thanks @enchoss

enchoss, 2018-02-06:
[image: 20180110_142540.jpg]

mundop, 2018-02-16:
Can the Sonoff running Tasmota + Smarthings Hub using this handler be controlled locally when Internet does down while you are in the same wifi network?

BrettSheleski, 2018-02-16:
This is a SmartThings device handler. SmartThings needs the internet to function. No internet --> no SmartThings

However, you should still be able to control your Tasmota device by other means, just not by SmartThings without an internet connection.

djtonyd, 2018-03-11:
hi all did anyone find out if the GPIO14 to GRND pin on a manual switch updates on the smart things app?

BrettSheleski, 2018-03-12:
Due to limits of SmartThings, the SmartThings app will not update automatically.

I've done some work with a new set of SmartThings device handlers and a SmartThings SmartApp to go with it. The SmartApp exposes an http endpoint which can be called externally to trigger the devices to update. I've done some testing with Node-Red where Node-Red listens to the mqtt topic of the Sonoff device and calls the http endpoint of SmartApp and it works.

However I need to do some more testing and I seldom have the time.

Knut91, 2018-05-01:
Hi, great device handler!

Is it possible to make the smartthings app react when i turn off the b1 with power?

Like this scenario. The bulb is on(also in the app) But i decide to turn it off with a pshyical switch so it loose all connection, because it got no power. But in the app it still is on, is it possible to make a command or something to send to smartthings app when it loose connection?

BrettSheleski, 2018-05-01:
I'm fully aware of the scenario you're talking about.  This is mainly due to a limitation with the way SmartThings works.

I built a SmartApp which exposes an HTTP endpoint which can be used to trigger status updates in situations like you describe.

This can be used with something like Node-Red which could listen to MQTT traffic for your Tasmota device(s) and call the SmartApp endpoint.

This process is not fully tested yet and in a state of (very slow) development.

Knut91, 2018-05-01:
Thank you so much for that quick answer.

As i am a really amateur, i will try to read me up on the node-red And i will try this out!
Do the node-red need another device to run it all the time or is it only on the web?

BrettSheleski, 2018-05-01:
I actually misread your initial question.  

If the device is off, meaning there is no power going to the sonoff device at all, I don't know of a good way of handling that within SmartThings.

andyjenkinson, 2018-05-02:
Obviously if you cut the power to the sonoff there is no way for it to send a message to SmartThings. So the only way to do this would be to constantly poll for the device from SmartThings and set the status to (though it should really be set to disconnected).

Brett, I too am interested in local switches being pushed to SmartThings though. I really would prefer it to happen via a direct HTTP call from the device though. I will have to modify the firmware.

My use case is actually for the T1 devices, and those devices really do need to update the handler when the button is pressed. I would also like to configure one of the  buttons to control a different SmartThings device instead (one of the lights in the room is a zigbee colour light) but that's just my own particular peculiarity.

BrettSheleski, 2018-05-02:
Check out my SmartThings SmartApp (https://github.com/BrettSheleski/SmartThingsPublic/tree/master/smartapps/BrettSheleski/tasmota.src).

Currently the SmartApp works.  However SmartThings requires going through OAuth to get a valid token (which needs to be passed in the HTTP headers) in order for requests to be made to the SmartApp.

If someone can get this working and provide instructions, feel free to make a pull request on my repo with the instructions in the markdown document (https://github.com/BrettSheleski/SmartThingsPublic/blob/master/smartapps/BrettSheleski/tasmota.src/readme.md).

I may create a simple HTML page which would do the OAuth handshake and obtain the info needed.  From there one could make the calls however they like.  One solution that I had working is create a flow in Node-Red to listen to MQTT traffic, and make HTTP calls to the SmartApp whenever .

I think it would be awesome if the Sonoff-Tasmota firmware were modified to call this directly.  This would eliminate needing external tools (like Node-Red).

Jacquesvdm, 2018-07-29:
Hi, I'm really keen to get a few sonoff devices going with the latest firmware and smartthings, when I click on your link to the device handler I get a 404 error, do you have an updated link to the device handler code please?

BrettSheleski, 2018-07-30:
I had to rename stuff in my GitHub repo (has to be lower case) for better SmartThings integration.

I updated the link on the wiki, where else also need be updated?

Raymon23, 2018-08-17:
Does anyone know how to change the LED to be ON when the Relay is OFF?

tuffelh, 2018-11-25:
Hi Brett, thanks for your work!
Can you explain what i have to add in preferences "Power channel" to view the actual Energy Consumption?
Or did i misunderstood this setting?
BTW: I use a Blitzwolf SHP2 and added "Case 45" to your code. Works great so far. Except it does not show the energy consumption.

BrettSheleski, 2018-11-25:
"Power Channel" has nothing to do with monitoring energy consumption. It is used for devices that have several relays attached to it (like the 4CH) where each "Power Channel" corresponds each output.

Please submit a pull request for "case 45" and corresponding handler needed.

In order to get energy monitoring working "properly" one would need to create a new SmartThings child Device Handler and implement the energy monitor capability. Doing so will however would still require the Tasmota device to send telemetry to SmartThings, which currently requires the help of Node-Red.

tuffelh, 2018-11-26:
Ah thanks for clarification. Don´t mind the energy monitoring yet ;)
I have created a pull request...

davidjac, 2018-12-10:
Links appear to be dead?

ascillato, 2018-12-10:
https://github.com/BrettSheleski/SmartThingsPublic/blob/master/devicetypes/brettsheleski/sonoff-tasmota.src/sonoff-tasmota.groovy

davidjac, 2018-12-17:
Many thanks, is this not the old one though. I was trying to find the version that feeds back "real" status.

[41 more replies]