arendst/Tasmota #657: Large shutter control board based on nodemcu, best way to do it with Tasmota?

neographikal, 2017-07-28:
I made the following setup:

[image]

with a nodemcu controlling all the relays. They work in pairs to control one shutter. I've tried espeasy, but the rest of my infrastructure is based on tasmota. Is there any way to do this with tasmota, what's the right way? Defining a custom type? I see only 4 entries for relays :)

davidelang, 2017-07-29:
unfortunantly it's hard to know how things are wired from the photo (especially 
without knowing what the various pins are)

Tasmota only supports 4 relays, look in sonoff_templates.ino for how things are 
defined.

David Lang
k

neographikal, 2017-07-29:
Each of the GPIO pins corresponds with one relay, high = relay on, low = relay off. The AC side has been wired so that if I would make a mistake programming the board, I can't short the motor of the shutter (only one side can be powered).

It would be nice to support a kind of combined relay, in that two outputs generate one "device". You would tag D0 and D1 with the same "SHUTTER_1" and Tasmota would expose a shutter device with a low/high attribute. Do that times 5 (max number of outputs) and it could work. Also usable for the Sonoff Dual/4CH to control shutters.

closed 2018-05-07 by stale[bot]