arendst/Tasmota #663: Sonoff POW enhancement

mkh595, 2017-07-30:
Hello,
would be possible that Sonoff POW reports also time in which was relay ON? E.g. TimeOnTotal, TimeOnToday, TimeOnYesterday.
I am using POW to control and monitor water pumps, compressors and UV lamps and they required after specific working time service or change of some parts.

davidelang, 2017-07-30:
since a power cycle will cause the device to loose track of everything that's 
happened so far, it's far better to do this sort of thing on your home 
automation server, or by looking at the logs in a syslog server

mkh595, 2017-07-31:
I am not using home automation, just only button on POW or Web interface, that's enough.
Calculation should be at source if possible because any failure with additional involved components (WiFi, mqtt, HA) cause inconsistency at HA because device can be still controlled manually.
Why should be power cycle problem? Energy Total and Energy Yesterday is also not lost.
My intention was:
- when relay goes ON, save millis() to rct memory
- when relay goes OFF, calculate time and update Totals, save 0 to rct memory
- periodically e.g. each 10 min update Totals and save new millis(). In case of problem we lost max 10 min which is not problem
- when there is unexpected reboot due to exception or power lost, we have last millis() stil in RTC memory if relay was on or 0 if relay was off

davidelang, 2017-08-01:
Energy yesterday is stored in flash at midnight, all totals for the current day 
are lost if the device reboots.

There is very limited space in these devices, so the code put in needs to be 
things that cannot be done reasonably elsewhere, and writing to flash frequently 
will wear out the flash, so we want to avoid doing so. The sonoff writes to 
flash when it's toggled, or at midnight for the yesterday values.

khcnz, 2017-08-01:
I think the power values are saved to rtc - which as far as i know should survive a reboot (but not survive mains power disconnection)

mkh595, 2017-08-02:
Today is my POW permanetly rebooted, more than 200 reboots. Totals are not lost.

00:00:10 MQTT: home/tele/home214/INFO1 = {"Module":"Sonoff Pow", "Version":"5.3.0", "FallbackTopic":"ESP_0767A2", "GroupTopic":"sonoffxxx"}
00:00:10 MQTT: home/tele/home214/INFO3 = {"Started":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x4020c65c epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000004 depc:0x00

Btw. after 2 minutes is Uptime changed from 0 to 1 hour

00:01:54 MQTT: home/tele/home214/STATE = {"Time":"1970-01-01T00:01:54", **"Uptime":0,** "Vcc":3.182,
00:01:54 MQTT: home/tele/home214/ENERGY = {"Time":"1970-01-01T00:01:54", "Total":8.670, "Yesterday":1.999, "Today":1.927, "Period":0, "Power":0, "Factor":0.00, "Voltage":0, "Current":0.000}
00:02:01 MQTT: home/tele/home214/UPTIME = {"Time":"1970-01-01T00:02:00", **"Uptime":1**}
00:02:07 MQTT: home/tele/home214/STATE = {"Time":"1970-01-01T00:02:07", **"Uptime":1**, "Vcc":3.182,

```

closed 2018-05-07 by stale[bot]