arendst/Tasmota #664: Help Please ESP-12F

pedroOliveiraptc, 2017-07-31:
Hello friend, have some one flash it and use it with success the ESP-12-F ? i lost hours and hours around it and cannot make it work :/
It works manually but don't work on mqtt.

http://www.ebay.com/itm/WiFi-Control-Pa ... 2371517466?

[image]
[image]

davidelang, 2017-07-31:
describe what's happening a bit more.

you flash it. Does it then connect to your network? can you reach the web page 
to configure it?

what happens when it tries to connect to your mqtt server?

is there any information in the logs? (have you increased your log detail to 
"more debugging"?)

k

stefanbode, 2017-08-05:
I assume you do not have enough Flash space and/or your firmware is too big to fit two times into the -F version (1MByte). The serial flash works always. The OTA Flash required at least double size of the firmware as the flash to store the new version until it reboots. The SONOFF and also the early ESP versions are limited by 1MB. This creates trouble in 80% of the cases. Not easy to solve here.

jeylites, 2017-10-04:
@pedroOliveiraptc 

in case you're wondering why this did work out, check the link below

[Yewelink/Tuya WiFi Touch wall light switch](https://www.letscontrolit.com/forum/viewtopic.php?f=5&t=3244)

closed 2018-05-07 by stale[bot]