arendst/Tasmota #665: GPIO 14 problem - Unstable state

ronwar, 2017-07-31:
I'm facing a problem , which might be caused by pin14
not stable state.It happens in all my Sonoff basic module.
Using GPIO14 and com for switching on and off connected to the
mechanical switch,the module strats to switch on and off
by it self (from time to time).
Did someone solved the issue,and is it possible to solve it
through change in code ?

arendst, 2017-07-31:
A 10k resistor between 3V3 and GPIO14 might open up a whole new world....

ronwar, 2017-07-31:
Thanks for quick response.I've already read about using pull up resistor,and your answer streagthen
this direction.

ionciubotaru, 2017-08-01:
Suggestion:
A parameter to choose between INPUT and INPUT_PULLUP will save a resistor for every switch.

arendst, 2017-08-01:
@ionciubotaru See line 2655 in sonoff.ino. Doesn't always seem to work......

gajotnt, 2017-08-02:
After adding resistors to sonoff, i had them always eventually failing... :(

droneando, 2017-08-17:
Hi:
I am having the same problems that described here when I connect a physical switch to GPIO14 to a Sonoff Basic, either if I configure it to control the relay or to send a mqtt topic: the device randomly interpret changes in the switch without there being any!!
I discovered that by connecting the switch in normally closed mode (GPIO14 connected to ground) the problems are solved. This works fine when you use a momentary switch(push switch), but when it is a common switch things get complicated again when switching ON (GPIO14 disconnected from ground).
Another problem is that the momentary switches usually only have two contacts in normally open mode, and this solution can not be used.
I have also discovered that by setting SwitchMode to 6 (PUSHBUTTON (Normally 0, 1 = toggle, Hold = hold)) the device randomly detects the HOLD position, which may be the reason for the malfunction. However, this solves the problem by doing nothing when HOLD.
It is not the most elegant way but it can be useful while modifying the code to solve the problem.

altelch, 2017-08-17:
Hi,

do you use a pullup for GPIO14? Look at 
https://github.com/arendst/Sonoff-Tasmota/wiki/Control-a-Sonoff-using-a-remote-button

Sounds to me like your GPIO14 gets floating if not grounded what gives 
your random behaviour.

Greetings,

      Heiko.

droneando, 2017-08-17:
**@altelch** The next thing I'll try will be the pullup resistor, but I'm afraid it will not work as **@gajotnt**  said...
Will keep you informed

droneando, 2017-08-17:
It seems that pullup resistor has solved the issue!!
Thank you all for your support!!

davidelang, 2017-08-17:
wire in a pull up resistor, and consider adding a capacitor in parallel to your 
switch if the wires are long or the resistor doesn't solve the problem.
k

ronwar, 2017-08-30:
What value of capacitor ?

davidelang, 2017-08-30:
On Wed, 30 Aug 2017, ronwar wrote:

> What value of capacitor ?

it's not critical, any small one should do (typically in the pF range)

ronwar, 2017-08-31:
I still face arbitrary on/off after wiring pull up of 10K (although longer period ) , I will try to add capacitor.

albukrek, 2017-09-03:
@ronwar,
I had the same issue. In my case, there were more than one topic ("cmd/<topic>/POWER1" and cmd/<topic>/power1"). 
In my case after this thread and others I have understood that there is  a need for  a resistor and capacitor.

closed 2018-05-07 by stale[bot]

Caswell1000, 2019-08-12:
> A 10k resistor between 3V3 and GPIO14 might open up a whole new world....

I need to try this, my cover states are suddenly erratic after working will without a pull-up resistor, GPIO14 is only used to report states buy it always indicates "OPEN" now.