arendst/Tasmota #666: Upgrade problem

gemitro, 2017-07-31:
Hι
I have problems when I try to upgrade my sonoffs to v5.5.0
I succeeded in upgrading them after more than ten (10) attempts for each device.
When the upgrade failed in "information" I was reading: **"Reset reason  Blocked Loop"**.
Thanks

0n3-70uch, 2017-07-31:
Hi,

same for me with this error:
> Failed Update error: ERROR[6]: Stream Read Timeout

I use "1M no SPIFFS" and also tested a reset before... Flashing Sonoff-minimal.bin is also impossible.

Greetings

arendst, 2017-07-31:
Blocked loop happens if the network takes too long to get acknowledges trying to retrieve the firmware. Does happen to me if I update 20 devices at once from my local webserver,

Stream Read Timeout means there is a network problem and it is hard to get your firmware from the internet.

Check your networks.

0n3-70uch, 2017-07-31:
Hi, I already checked the network and restarted everything (router and so on). I also reduced the distance between one Sonoff and the wifi access point... no success. The signal quality is round about 95-98%.

I notice the same with many devices therefore I started the upgrade procedure for every device manualy.

Seems that I need to flash the devices via serial.

davidelang, 2017-08-01:
are you trying to download the update from the Internet or from a local server?
k

ionciubotaru, 2017-08-01:
Upgrade rules:
- wifi signal must be more than 70%
- perform one upgrade once (if all your sonoffs start the upgrade process in the same time, the web server cannot serve all the requests)
- if upgrade command fail, try to upload manually via web interface

0n3-70uch, 2017-08-01:
Hi and thanks for our help...
Six devices ignore the update. Some of theme answer with 
"Failed Update error: ERROR[6]: Stream Read Timeout"

These steps are tested:
- reduced the distance to a signal quality over 95%
- only one upgrade at the same time
- a restart, short break, start upgrade
- a restart, reset 2, short break, start upgrade
- webinterface upgrade (with upload)

Its a bit frustrating... I will upgrade via serial today.
Greetings

gemitro, 2017-08-01:
I tried both from the internet and from a local server, with the same results.
It's definitely better when I upgrade a device at a time, but even so I have to try two or three times.
I can not be sure, but I think the problems started with v5.3.0

ionciubotaru, 2017-08-01:
It's not a Tasmota problem since @arendst use the standard OTA library.
I recomand to fire the upgrade of one device at a time and read MQTT messages for Succes, otherwise retry upgrade until succed, then start the upgrade of the second device

0n3-70uch, 2017-08-01:
Yeah feel the same... since 5.3.0. I had no trouble with version 2.x or 3.x but since version 5.x its a bit "try and error". I only get this short "ERROR[6]" with active debug loglevel.

Fabi

pascalsaul, 2017-08-08:
I've also updated all my devices (~20) at the same time to various 5.x releases without issues every time.

0n3-70uch, 2017-08-09:
Hi, what sort of hardware do you use as server?

Greetings Fabi

arendst, 2017-08-09:
As @pascalsaul I also use a Virtual Machine (VM) on my Intel NAS running Debian Jessy.

The VM runs Debian Jessy too with Mosquitto, Apache 2.4 and Domoticz slave. A Raspberry Pi is my Domoticz master.

0n3-70uch, 2017-08-09:
Ok so you are updating the devices via the powerful vm... I will install an apache system on my ubuntu destop pc and make another test. At the moment I use the raspberry pi with apache server for updates.

pascalsaul, 2017-08-09:
https://www.zotac.com/us/product/mini_pcs/zbox_c_series (I use the older CI323 nano). These boxes got more power to handle the requests. 

@arendst Maybe you could built-in an option to schedule the upgrade with a random generated delay within a 5 minute timeframe to minimise the load. So you can spread the load from the devices over 5 minutes.

0n3-70uch, 2017-08-09:
Thanks for the link. At the moment I make Device Groups `Sonoffs_Group1` ... `Sonoffs_Group2` and so on. Every group with 5-10 devices. Later I upgrade the devices with the command `cmnd/Sonoffs_Group1/Upgrade -m "1"`. Not the best solution but working for me.

arendst, 2017-08-10:
The OTA upgrade routine contains some kind of retry mode which should cater for a busy webserver although in practice it probably ends up with a restart using the old version.

If it restarts using the old version you could try command ``cmnd/sonoffs/upgrade <new-version>`` in which case devices with the \<new version\> will ignore the update but the once with the \<old version\> will try to upgrade.

0n3-70uch, 2017-08-11:
Hi Theo,

thanks for the tip, this is a nice feature. Now I can upgrade the devices with an older version without searching for the correct ID...

Greetings Fabi

0n3-70uch, 2017-08-11:
Short update: I told you that some of my devices ignored the upgrade to a version after 5.4.0. Now I erased the flash with `esptool.py -p /dev/ttyUSB0 erase_flash` and flashed the firmware 5.5.0. I flashed the newer version 5.5.2 via ota after this procedure... working fine now. I will update this issue after some test.

Greetings Fabi

0n3-70uch, 2017-08-14:
I close this issue now... Seems that the upgrades working now.

closed 2017-08-14 by gemitro