arendst/Tasmota #668: Led2 is not working as it should

chukiki, 2017-07-31:
Hi,

I used the Mini ESP-M1 (esp8285 - https://www.banggood.com/Mini-ESP-M1-ESP8285-Serial-Wireless-WiFi-Transmission-Module-IoT-Compatible-With-ESP8266-p-1152979.html).
Flashed successfully with tasmota firmware.
I configured the following:
GPIO4   = LED1
GPIO14 = SWITCH1
GPIO12 = RELAY1
GPIO5 = LED2
GPIO15 = SWITCH2
GPIO13 = RELAY2

I get to see two toggle buttons on the webpage and can control using the commands page the POWER1 and POWER2, but, they both only control LED1 while LED2 is completely ignored.
So even if POWER2 is on and POWER1 is off still LED1 will be on.
Is it a bug or something that I'm doing wrong?

arendst, 2017-07-31:
It's a feature as most Sonoffs only have one controllable Led1. I never looked at the other leds. Just do as iTead and connect the led to the relay gpio...

chukiki, 2017-08-01:
Hey,
I would really appreciate it if the other leds would work.
I tried to connect one of the leds to RELAY2 and it did work, but only as long as I had one RELAY2.
Once I had two RELAY2, one for the led and one for the actual relay, it did not work.

closed 2018-05-07 by stale[bot]