arendst/Tasmota #669: i2c character LCD Display support

chukiki, 2017-07-31:
Hi,

I'm using ESP8285 with tasmota and really enjoying it.
Meanwhile connected two leds, switched and relay.
Also planning to connect a DHT22 and a LCD Display screen.
I really wanted to know if currently or in the furue there is or will be support in a i2c character LCD display like the following ([https://www.banggood.com/IIC-I2C-2004-204-20-x-4-Character-LCD-Display-Module-Blue-p-908616.html]).
I would really love it if I'll be able to send throw MQTT a 4 lines to be written to the LCD display.
Is there a way to do it right now? or planned to be?
Thanks a lot for all the good work!

arendst, 2017-07-31:
Once worked on it. Have one in use but the problem is that what I like someone else doesn't and then there have to be made code additions that someone likes but another doesn't so I have to satisfy three people already. There are millions of these individuals around all wanting something different involving more code and code space is scarce on a 0.5 M device.

Bottomline, I could implement a simple driver for a I2C screen with basic mqtt commands like LCDCOL, LCDROW and LCDTEXT but I doubt if it would be enough...

ionciubotaru, 2017-08-01:
I suggest to add a `user.ino` with a `user_lcd()` function editable by user. By default print `Hello word`, then the user can adapt the output to it's own needs. There are  2 lines displays, 4 lines and more other. First user will need the humidity and temperature, the second all 4 temperature sensors, the third the switch state .....

Better allow each user to configure output. A list of available useful variables (temperature, relay state, switch) with a short description at the beginning of the file will help.

chukiki, 2017-08-01:
I agree about few configurable options, while the basic and most important is the one that arendst suggested. Maybe I would extend it a little to support an mqtt command of setline <index>, <text>.
Regarding what ionciubotaru said, second step I would allow to configure 2 line\4 line and what to be shwon, while basic is if connected to wifi or not and ip adress and connected device state (power1: on... temperature, humidity, etc).
Regarding the space issue, will it be possible to supply several build like with the BE_MINIMAL
So also build that fits people who has or upgraded to 4MB flash and contains more options?
My router's firmware open-source project (http://tomato.groov.pl/) does that to supply several different builds depending on the features that you want\you router brand and memory you have in it...

PeggyFree, 2017-08-01:
This feature is the best idea I've read so far. Even if it keeps basic with
minimal options.

chukiki, 2017-08-01:
Haha, which one of them? I had several out there :)

kruebsam, 2017-08-02:
I´ve 'by default' upgraded all of my S20 and alikes with a 4 Mbit chip, thus space is currently no problem (at least for me). I would also second chukiki 's comments: Why not leave it up to the user to decide: Have a standard version which works without any hazzle on a plain S20 and alikes or have (memory-)upgraded devices that allow additional bells and whistles, such as LCD-support.

liorsu, 2017-08-02:
Votes for that idea as well. Already got the 4MB chips and now waiting to upgrade to it. As well, bought some built in 4MB ESP boards.

stefanbode, 2017-08-03:
I think the best option is to enable the capability through a compiler Option. That makes is suitable for all users to decide how to work and what to compile. My SONOS Relays get a different compile than the nice WEMOS with 4MB. Anyhow, quite soon we will also hit limitations in the RAM. And this is not easy to overcome on the EPS8266. It is just a mini computer. The new ESP32 will add much more capabilities and then i2c displays are no more an issue. Currently, we should accept the limits.

closed 2018-05-07 by stale[bot]