arendst/Tasmota #670: S20 - Expansion by AM2301 Temperature &  Humidity

FabianHM, 2017-07-31:
Simple solution to expand the S20 by an AM2301 module.
[image]
[image]
[image]
[image]

jat80, 2017-08-06:
What pins go to what from the S20 to the audo jack?

arendst, 2017-08-09:
Thnx. Closed issue and add link in Wiki.

closed 2017-08-09 by arendst

jat80, 2017-08-11:
@arendst I still don't know how to configure this?

arendst, 2017-08-11:
For others:

[image]

jat80, 2017-08-12:
@arendst It's the wiring for the jack to correspond with the socket that would be the issue for most people.

FabianHM, 2017-08-12:
You need only connect 3 cables!
AM2301 red to S20 VCC (3.3V)
AM2301 black to S20 GND (Ground)
AM2301 yellow to S20 RX or TX (GPIO 1 or 3)

jat80, 2017-08-12:
@FabianHM how would someone wire the 3.5mm jack is what i'm saying.  the above is obvious but if someone wants to make there own jack to the sensor themselves, which pins correspond with the jack pins - jack to jack socket.

FabianHM, 2017-08-12:
Connect the audio cable with 3.5mm stereo plug with the Jack socket.
Now simply measure the pin assignment of the socket with a measuring device. The allocation of the bushes are different in the case of some manufacturers.

CaptInsano, 2017-09-24:
Is any pull-up resistor needed between VCC and DATA?

I am getting elevated Temp measurements from my AM2301s (I have tested with 3 different AM2301s) on different S20 sockets and I am trying to debug why.

I have collected a DS18B20 to the other S20 pin to compare readings, see below:

[image]

[image]

Any ideas?