arendst/Tasmota #672: inconsistent mqtt publish

holgerimbery, 2017-08-01:
Hello,
I configured my devices to use %topic%/SON_xxyyzz/%prefix%/ as full topic.
Everything beside prefix: "stat" is working as expected, but the prefix stat is published as:
relay switched via mqtt
```
07:30:01 MQTT: stat/SON_CCC2A8/RESULT = {"POWER":"OFF"}
07:30:01 MQTT: stat/SON_CCC2A8/POWER = OFF
```
relay switched via button on device
```
07:31:11 MQTT: sonoff/SON_CCC2A8/stat/RESULT = {"POWER":"ON"}
07:31:11 MQTT: sonoff/SON_CCC2A8/stat/POWER = ON
```

Is this a bug, or a configuration issue ?
I'm using v.5.5.0 on S20.

Regards,
Holger

arendst, 2017-08-01:
#631

holgerimbery, 2017-08-01:
Theo you are right, stupid me - this is a configuration issue.
*close*

closed 2017-08-01 by holgerimbery