arendst/Tasmota #674: sonoff POW not work with main power after flash  Tasmota 5.5

fparodo, 2017-08-01:
Hello,

I Flashing TASMOTA with PlatformIO on my sonoff POW but when disconnect the ftdi adapter and connect the mains the sonoff POW dosen’t work . Can you help me ?

with the ftdi adapter work fine
please let me know

ionciubotaru, 2017-08-02:
Try to measure the 3V3 output, to check if the power source is working fine
If ESP works via FTDI and not connected to mains this means you have a power source failure

jetema, 2017-08-02:
Check as well if you have 5V at the ouput capacitor ( D5 terminals ), the micro power supply gets first this voltage ( 5V ) to get 3V3 on U2 regulator and power on both, the ESP8266 and flash memory.
By the way the relay works with 5V, not 3V3  
[image]

pascalsaul, 2017-08-05:
I had this also with a TH16..., bad luck? I'll check D5 also for the 5V.

PeggyFree, 2017-09-09:
Just had the same problem with my brand new pow. 
Flash using FTDI : OK
Test and calibration with a 100w bulb : OK
Definitive installation : Fail. No LED, nothing.
Using a multimeter, I see no 3.3v, no 5v, always 0v.

jetema, 2017-09-12:
Maybe is not worth trying to repair your sonoff, anyway in case you love electronics ( As me :-), here you have the schematic ( power suplly area ) If you decide to try GOOD LUCK
[image]
 ;-)

PeggyFree, 2017-09-21:
Back to my previous comment. It's the third time it happens with different types of sonoffs : flash OK, test OK, real use : fail. No LED light, no data. Some hours later : working again like a charm. 

There must be a kind of resettable fuse.

ionciubotaru, 2017-09-21:
I have the same problem, Sonoff POW starts working after 5 minutes when is connected to mains. 
The problem is in the power supply, not in esp8266

jetema, 2017-09-21:
If you want to know more you also could try powering on externally the sonoff, It just needs 5V ( any USB mobile charguer will do ) , so is one way to check if it still does strange things as well ( or not ) with DC external supply.
In case you decide go on, disconnect the mains from sonoff, then connect +5V on  D5 terminals ( blue mark positive, D5 is a zenner diode ) now you can check if there's any delay or not in data or led.

closed 2018-05-07 by stale[bot]