arendst/Tasmota #675: using Mqtt Fulltopic wildcards

Sunrise17, 2017-08-01:
1st ex.;
Fulltopic of Device: Home/Kitchen/BenchLamp/sonoff03/cmnd/

When i subscribed to “tele/#”
RESULT: Nothing

2nd ex.;
Fulltopic of Device: cmnd/Home/LivingRoom/CornerLight/sonoff03/

When i subscribed to “tele/#”
RESULT: tele/Home/LivingRoom/CornerLight/sonoff03/LWT  “Online”

In second example, it works well but i want to control devices in following below, i have not received any response yet after try. 

cmnd/Home/LivingRoom/power “on”
or
cmnd/Home/power “on”

I will be glad if you can guide for this issue.
	
Thanks for your effort,

davidelang, 2017-08-01:
look at the logs (increasing log details if needed), it looks like you are 
changing the fulltopic, but not changing what you are trying to subscribe to.

looking in the logs, you will be able to see what the tele messages are and can 
format your subscriptions to match.
k

Sunrise17, 2017-08-01:
Regarding to 1st example, tele messages in the log in this format Home/Kitchen/CeilingLight/sonoff03/tele/STATE , but i have not found how to subscribed it #/tele or else??

and 2nd question is, if we can succeed to subscribe, can we control switches room by room or home by home? such as (Home/Kitchen/cmnd/power "on"  or Home/cmnd/power "off")

joecotton, 2017-08-01:
The # wildcard can only be used at the end of a subscription topic. Since your fulltopic structure puts 'tele' closer to the end of the string, then your wildcard would have to be

`Home/Kitchen/CeilingLight/sonoff03/tele/#
`

Regarding your 2nd question: The TASMOTA firmware allows for a "Group" topic in addition to the primary topic. By default, this group topic is 'sonoffs' and by having multiple devices subscribe to this same topic, you can control the whole group of them.

To control by "room", just set the GroupTopic to your room specifier.
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#mqtt-specific

davidelang, 2017-08-01:
On Tue, 1 Aug 2017, Sunrise17 wrote:

> Regarding to 1st example, tele messages in the log in this format Home/Kitchen/CeilingLight/sonoff03/tele/STATE , but i have not found how to subscribed it #/tele or else??

I think you would need to do something like #/tele/#, but you may need to 
enumerate the levels

> and 2nd question is, if we can succeed to subscribe, can we control switches room by room or home by home? such as (Home/Kitchen/cmnd/power "on"  or Home/cmnd/power "off")

look at grouptopic for controlling several devices with a single command.

Sunrise17, 2017-08-01:
I found the way to subscribe for  Home/Kitchen/CeilingLight/sonoff03/ which is " +/+/+/+/tele/# ", however with same method i have not succeded to switch on/off in group as like +/+/+/+/cmnd/power "on". In addition, group topic has some issue, it does not work for topics having full topic as like in this example.

Sunrise17, 2017-08-01:
here is really guide for single level (+) hierarchies. http://www.tigli.fr/lib/exe/fetch.php?media=cours:tutorial_mqtt_mit_2015_2016.pdf

mkh595, 2017-08-02:
Wildcards can be used only for subscribtion, not for publishing.

Sunrise17, 2017-08-02:
So, there is no possibility to control only rooms or homes in this format, group topic is not solving the issue.
cmnd/Home/LivingRoom/CornerLight/sonoff03/power
cmnd/Home/LivingRoom/power
cmnd/Home/power

closed 2017-08-02 by Sunrise17

davidelang, 2017-08-02:
On Tue, 1 Aug 2017, Sunrise17 wrote:

> So, there is no possibility to control only rooms or homes in this format, group topic is not solving the issue.
> cmnd/Home/LivingRoom/CornerLight/sonoff03/power
> cmnd/Home/LivingRoom/power
> cmnd/Home/power

you could make a livingroom group topic for all lights in that room, but you 
could not also have a third topic for the whole house.

you keep repeating that group topic is not solving the issue, but you haven't 
said what it's failing to do.

David Lang