arendst/Tasmota #676: Open/flash Sonoff B1 Light Bulb

qorron, 2017-08-01:
just got one, but I can't access the serial connection pins yet.
at least without breaking something.
the base seem to contain the power regulator, everything else is in the top.
it can be unscrewed from the big heat-sink but that doesn't help because the 2 parts are still attached to one another via the cable.

to access the electronic the plastic cover has to be removed. has anyone successfully accomplished that without breaking it?

once I accomplish that, I'd like to help testing the tasmota firmware for the B1

mredone, 2017-08-02:
Yes. I open B1 Bulb.
I use my work gloves and twist plastic led clover and pull one side and cover pop out.

Maybe i can help testing tasmota firmware on that Bulb.

timtimsson, 2017-08-04:
[image]

arendst, 2017-08-05:
@timtimsson I understand that you managed to flash the device with Tasmota v5.5.0. What happens if you select module Sonoff LED? Are you able to control both white LEDs? Do not expect color LED control as it has not been built into Tasmota yet.

timtimsson, 2017-08-05:
@arendst  ...yes i use the Sonoff Led Modul, but nothing :(
: Color FF00, FFFF, 00FF + Power ON + Dimmer 100
we need the layout....
if i flash my own dump back, it works again with ewelink....

timtimsson, 2017-08-11:
[led_10w_5.pdf](https://github.com/arendst/Sonoff-Tasmota/files/1217029/led_10w_5.pdf)

timtimsson, 2017-08-11:
[image]

arendst, 2017-08-11:
@timtimsson I hope it's the correct schematics...

My B1 is on it's way and should arrive anytime now.

timtimsson, 2017-08-12:
@arendst : my B1 has a 8285...
my9231 - maybe you can use library my9291?

arendst, 2017-08-12:
@timtimsson I just uploaded experimental support for Sonoff B1 in the development branch.

Please give it a try.

- Select Module Sonoff B1
- Change color with command Color RRGGBBCCWW so it needs 10 hexadecimal values. I'm not sure the leds are being controlled in the correct order (if any led is being controlled at all).

Hue support is not available yet.

TBC

timtimsson, 2017-08-12:
... 17:11:44 MQTT: /ESP_Bu1/stat/RESULT = {"POWER":"ON", "Dimmer":64, "Color":"A3A3A3A3A3"}
but nothing, all leds are dark....

arendst, 2017-08-12:
Pity. You'll have to wait for my B1 to arrive and I can do some testing.

BTW what is the result of ``color A300000000``?

timtimsson, 2017-08-12:
17:24:24 CMND: color A300000000
17:24:24 MQTT: /ESP_Bu1/stat/RESULT = {"POWER":"ON", "Dimmer":63, "Color":"A000000000"}
- all leds dark....

arendst, 2017-08-12:
What if you enable interrupt control by removing the slashes in file xdrv_snfled.ino from lines 80, 118, 128 and 156?

timtimsson, 2017-08-12:
.. i remove all slashes: but nothing same result, dark...
17:33:59 CMND: color A300000000
17:33:59 MQTT: /ESP_Bu1/stat/RESULT = {"POWER":"ON", "Dimmer":63, "Color":"A000000000"}

timtimsson, 2017-08-12:
i hope the circuit plan is not wrong - i receive it from sonoff direktly ???

arendst, 2017-08-12:
Again, pity.

Have to go. I'll try when I have mine.

timtimsson, 2017-08-12:
ok - thanks a lot...

FabianHM, 2017-08-12:
After I read this I have ordered a Sonoff B1...it takes normaly 20 Days and I hope tasmota is supported then

arendst, 2017-08-13:
Just made a new branch called ``Sonoff-B1`` with some changes that might work on the bulb. 

Pls give it a try.

Still waiting for mine to arrive.

timtimsson, 2017-08-13:
... {"StatusFWR":{"Program":"5.5.2d", "BuildDateTime":"2017-08-13T11:54:00", "Boot":6, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
CMND: color A300000000
MQTT: /ESP_Bu1/stat/RESULT = {"POWER":"ON", "Dimmer":63, "Color":"A000000000"}
... but nothing, everything is still dark  :cry:

arendst, 2017-08-13:
Try color 0000A00000

timtimsson, 2017-08-13:
11:09:46 CMND: color 0000A00000
11:09:46 MQTT: /ESP_Bu1/stat/RESULT = {"POWER":"ON", "Dimmer":62, "Color":"00009E0000"}
...dark

arendst, 2017-08-13:
Thnx for trying.  Back to my shed...

timtimsson, 2017-08-13:
yes of course...If you have any ideas, please share it, i will test it again..

arendst, 2017-08-13:
Did update with another Module (Sonoff B1 b) using different GPIO pins (but seem to be more used with type of bulbs)

Pls give it a go.

timtimsson, 2017-08-13:
no, gpio 13,15 also doesn't work :(
i can't find my9231 on the top of my platine - how can i find the right pins?

arendst, 2017-08-13:
I've seen pictures where they were able to remove the platine from the shell and showed the power supply underneath. Perhaps if you are abvle to remove it too you can see underneath to see the my9231 or my9291 chip.

timtimsson, 2017-08-13:
pity, i can't remove the platine on the top with a screwdriver and the platine inside is coverd with a plastic tube....
if i remove the tube, i have no chance to isolete the platine after this, because the wire are on the back from the topside platine...
[image]

[image]

timtimsson, 2017-08-13:
ok...

arendst, 2017-08-15:
Received mine today. Made some progress: 

[image]

Still not there although the schematics seems to be OK as I can control "some" led.

arendst, 2017-08-15:
Just uploaded (at least with me) working Sonoff B1 version in the Sonoff-B1 branch.

Pls give it a try.

timtimsson, 2017-08-17:
very very good!  :sunglasses: :thumbsup: :heart_eyes:

mredone, 2017-08-17:
Hi. My gpio0 pad is broken. Can i find it anywhere?

timtimsson, 2017-08-17:
@mredone 
maybe you can try this - i don't know:
[image]

mredone, 2017-08-17:
Now i search for right leg on chip but i cant find clear photo with info where is gpio0

timtimsson, 2017-08-17:
@arendst 
only a question: if i take the color slide in the middle {"POWER":"ON", "Dimmer":100, "Color":"0000007F7F"}, then is this not 100 -> FFFF is 100
but not so important for me, because i use mqtt

timtimsson, 2017-08-17:
@mredone 
https://www.itead.cc/wiki/images/b/b5/0a-esp8285_datasheet_en.pdf
dot is left/up - gpio0 is bottom from right the 2nd...

mredone, 2017-08-17:
Code upload but b1 not start. No HTTP server. No logs via serial

arendst, 2017-08-17:
@mredone flash mode dout?

https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips

mredone, 2017-08-17:
Damn. On 8266 i have DOUT. But when i chance to 8285 Arduino Ide not show that option. Damn!

arendst, 2017-08-17:
@mredone less damn as the esp8285 selection automatically selects dout but it might use the wrong linker file.

Why not stay at your esp8266 set to dout. Other than fixed dout there is no change in selecting esp8285 but changing the settings in the ide might default to settings you do not want.

Bottomline: always use esp8266 with flashmode dout and linker file 1M no spiffs.

rajil, 2017-08-26:
Can somebody post picture/instructions on where the header needs to be installed on the PCB?

ThomDietrich, 2017-09-01:
@arendst some pictures for the wiki would be quite helpful

FabianHM, 2017-09-01:
Today my B1 arrvied and I need help to open it.
[image]

How Can I open the acrylic cap? 
Is the Cap fixed with a thread?

qorron, 2017-09-01:
screw it back on and pop it off with controlled brute Force.
just don't cycle the power 3 times in a row. because then the power supply fries.

richrd, 2017-09-03:
@qorron what do you mean by frying the power supply? :thinking: I'm going to get some B1 bulbs at some point and definitely don't want to fry them.

rajil, 2017-09-03:
The pads on this thing are small so soldering on this is difficult. I dont see any button so how do you put this into flash mode?
My flash upload is simply failing now.

HellfireZA, 2017-09-04:
Is it not possible to write the Tasmota firmware to the bulb which is connected to the network via "Over the Air"?

ageorgios, 2017-09-07:
Any guide how to flash the B1?

badgerhome, 2017-09-09:
I would like to know the same please, how do I flash the B1? Thanks :)

msth, 2017-09-12:
a howto would be very helpful

janvde, 2017-09-12:
I've succesfullly flashed my Sonoff B1. It's just the same as with most sonoff devices, like the s20. Just connect the programmer to the 4 pins on the left. The only difference is that you need to connect the GPIO0 with the GND when attaching the FTDI USB-TTL programmer. After that you can disconnect those and upload your code.

eggfat, 2017-09-13:
Could someone make a step by step or video please?
Thanks.

ageorgios, 2017-09-13:
any video disassembling the top part?

ThomDietrich, 2017-09-13:
It is amusing to follow this thread. Nowadays *everyone* has a smartphone with a camera and internet access with automatic cloud storage. How can it be such a fuzz to upload a picture so others can profit from ones experience!?

There are device specific articles in the Tasmota wiki (e.g. [Sonoff Dual](https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-Dual)), it would be astonishing if someone could create such an article for the B1. I would do it myself if I had such a device.

ageorgios, 2017-09-13:
any ideas flashing the B1 without soldering?

davidelang, 2017-09-13:
I don't solder wires to anything, I use spring loaded pogo pins and hold them in 
place.

you can do the same thing with bare wire (as long as you have something large 
enough to be reasonably stiff)

arendst, 2017-09-13:
Some people managed to use sonota with their devices. I don't know if the B1 would work though.

Take a look in the wiki regarding upload and follow the link to sonota.

It should allow for uploading tasmota from itead ewelink.

HellfireZA, 2017-09-15:
When reading what @janvde  said, in regards to the fifth pin. I don't understand... slightly vague there, you say connect the GPIO0 with the GND, do you mean that there's one cable coming from the FTDI GND pin that splits into 2 connections to the bulb, 1 for GPIO0 and 1 for GND resulting in 5 points of contact on the bulb but only 4 pins being used on the FTDI?

ThomDietrich, 2017-09-15:
I don't own a B1 myself but from what I've read so far the method should be like follows. If someone could please confirm that, this thread can finally be put to sleep.

Just as for most other modules you need to connect the FTDI programmer just as explained under: https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation#serial-connection

Additionally one needs to connect GPIO0 with ground during power up. That is described here: https://github.com/arendst/Sonoff-Tasmota/wiki/Hardware-Preparation#bringing-the-module-in-flash-mode

Now the final and most important question is where one will find the 4+1 pins. As can be seen in the image by @timtimsson they are available on the PCB as testpoints. In the image they are on the top half and clearly marked. You need to access these by either soldering wires to them or by using pogo pins as suggested above.

[image]

Good luck.

[31 more replies]

closed 2017-09-16 by qorron